davidker/unisys #2: (rc != 0) is a double negative

davidker, 2016-03-02:
(rc != 0) needs to go, from everywhere

davidker, 2016-03-09:
Also "if (rc != 0)" is a double negative.  != zero is good style when
you are talking about the number zero or for strcmp() functions.

arfvidsonUIS, 2016-03-11:
Fixed all !=0 except those for uuid comparisons. Just pushed the branch with the name upstream-double-negative.

Here is the list of the ones I fixed marked with xxxx
(11:22 rootx@ubuntu unisys) > find . -name "*.h" | xargs grep "!= 0"                                                                   
./include/channel.h:    if (uuid_le_cmp(expected_uuid, NULL_UUID_LE) != 0) {
./include/channel.h:            if (uuid_le_cmp(guid, expected_uuid) != 0) {
xxxx./include/channel.h:            if (readb(&hdr->cli_error_os) != 0) {
xxxx./include/channel.h:    if (readb(&hdr->cli_error_os) != 0) {
xxxx./include/channel.h:    if (readb(&hdr->cli_error_os) != 0) {
(11:22 rootx@ubuntu unisys) > find . -name "*.c" | xargs grep "!= 0"                                                                   
./visorbus/visorbus_main.c:          (uuid_le_cmp(drv->channel_types[i].guid, NULL_UUID_LE) != 0) ||
xxxx./visorbus/visorbus_main.c:     if (rc != 0)
./visorhba/visorhba_main.c:     if (!wait_event_timeout(notifyevent, notifyresult != 0xffff,
xxxx./visorinput/visorinput.c:              int old_state = (test_bit(led, visorinput_dev->led) != 0);

davidker, 2016-03-11:
I want to keep this reopened until it gets into staging-next. Just in case we get yelled at by gregkh.

davidker, 2016-03-13:
Transferred two of the patches over to upstream-next. I have not transferred 2ab6c11 as that one conflicts with upstream.

davidker, 2016-03-25:
Greg accepted this into staging-testing so I'm closing this ticket.

closed 2016-03-25 by davidker