davidker/unisys #3: Eliminate vaguely named variables

davidker, 2016-03-02:
char s[99]
x
t

Take care of these.

gigatwo, 2016-03-08:
Is patch dd22ded11dc8a11026fe9f8dbcc711cd49794aae about the right general idea?

gigatwo, 2016-03-09:
e01bf1c4533ccfc0ca2db9df514bf239ef3f60ca should removes the "s[99]" bit.

gigatwo, 2016-03-10:
I ended up replacing 'p' with 'pos' and 'x' with 'shift' in client_bus_info_show. not sure if the first one was overkill.

gigatwo, 2016-03-10:
At this point, visorbus_main.c should be clean.

davidker, 2016-03-13:
Moved the patches over to upstream-next. Reworking a couple of the comments since they reference github issues.

davidker, 2016-03-13:
Are you planning on getting rid of the "rc" as well?

selltc, 2016-03-14:
FWIW, I finally got around to reading https://github.com/davidker/unisys/blob/staging-next/Documentation/CodingStyle, and it basically says this about temporary variable names:
```
LOCAL variable names should be short, and to the point.  If you have
some random integer loop counter, it should probably be called "i".
Calling it "loop_counter" is non-productive, if there is no chance of it
being mis-understood.  Similarly, "tmp" can be just about any type of
variable that is used to hold a temporary value.
```
So given that `i` is ok for a temporary int, I'm not sure why `s` isn't ok for a temporary string.  Perhaps we should just quote the CodingStyle text the next time we're called out for something like this.

davidker, 2016-03-25:
greg has accepted into staging-testing last night, so I'm closing this.

closed 2016-03-25 by davidker