davidker/unisys #5: Raw kobject -> Driver specific object conversion

davidker, 2016-03-02:
Needs discussion. What's a doable approach for this that still allows us to do small patches?

selltc, 2016-03-08:
Commit 28266c54e8159cd0d614da086a92f760fc65c62d addresses this by removing the only remaining usage we have of raw kobjects.  We no longer use /sys/devices/visorbus<x>/vbus<x>:dev<y>/devmajorminor/* for anything in the upstream, so this is safe to remove.  Suggested patchset headline:
`Remove last remaining remnants of raw kobject usage from visorbus`

davidker, 2016-03-11:
This got rejected into staging-testing due to it being behind upstream. Need to resend with it being in the series for messy gotos. #4

selltc, 2016-03-11:
I actually think we might be better-off submitting this BEFORE re-submitting the messy gotos (issue #4).  Reason is, I think it is possible that Greg might have a problem with us changing this much stuff in a v2 patchset.  Is this a valid concern, or am I worrying about nothing?

selltc, 2016-03-11:
I would be more-than-happy to rebase commit 28266c54e8159cd0d614da086a92f760fc65c62d on top of the latest staging-next if necessary.

davidker, 2016-03-11:
I'm changing the order of the patches in the upstream and reworking it. I'll be sending out an inspection soon. This will be the first patch in the messy goto series for visorbus.

davidker, 2016-03-22:
It is in staging-testing and no nasty grams. I'm closing this issue.

closed 2016-03-22 by davidker