davidker/unisys #6: visorchipset.c:669 remaining_steps_show() warn: XXX passing uninitialized > 'remaining_steps

davidker, 2016-03-08:
DanC's smatch run says:

drivers/staging/unisys/visorbus/visorchipset.c:669 remaining_steps_show() warn: XXX passing uninitialized 'remaining_steps

Needs a patch.

davidker, 2016-03-09:
drivers/staging/unisys/visorbus/visorchipset.c:542 toolaction_show() warn: XXX passing uninitialized 'tool_action' \
drivers/staging/unisys/visorbus/visorchipset.c:609 error_show() warn: XXX passing uninitialized 'error' drivers/staging/unisys/visorbus/visorchipset.c:639 textid_show() warn: XXX passing uninitialized 'text_id' \
drivers/staging/unisys/visorbus/visorchipset.c:669 remaining_steps_show() warn: XXX  passing uninitialized 'remaining_steps'

TurningWrenches, 2016-04-04:
The issue is repaired:

branch: passing_uninitialized_var_issue
checkpatch-cleared: yes
T710-1 verified: yes

selltc, 2016-06-09:
Greg committed this to staging-next 4/29:
* commit 3a56d700f60cd3cdc2c1429d107f0818555a5b22 - staging: unisys: visorbus: initialize variables

closed 2016-06-09 by selltc