davidker/unisys #8: int maxdevnodes = ARRAY_SIZE(dev->devnodes) / sizeof(dev-

davidker, 2016-03-08:
Should just be ARRAY_SIZE(...);

Look up ARRAY_SIZE for fun and see!

selltc, 2016-03-09:
Fantastic find David.  Fortunately, the bad line was removed by commit 28266c54e8159cd0d614da086a92f760fc65c62d for issue #5, which David is in the process of submitting upstream, so I think we can close this.

I looked for this line in our legacy drivers, but couldn't find it there.  Indeed I found our legacy drivers have the correct code:
`int maxdevnodes = sizeof(dev->devnodes)/sizeof(dev->devnodes[0]);`
, so it looks like this was an upstream-only snafoo.

closed 2016-03-09 by selltc

davidker, 2016-03-09:
Don't give me credit for finding it, you can give Dan C the credi for it.