davidker/unisys #10: checkpatch errors, warning, and checks:

davidker, 2016-03-09:
kershnda@USTR-ERL-4458[28013.kernel]:~/kernel/github/unisys$ find drivers/staging/unisys -name '*.c' -o -name '*.h'  | xargs scripts/checkpatch.pl  --strict --terse -f
drivers/staging/unisys/visorbus/visorchipset.c:536: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:571: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:602: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:632: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:663: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:1232: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:1453: WARNING: char * array declaration might be better as static const
drivers/staging/unisys/visorbus/visorchipset.c:1603: WARNING: char * array declaration might be better as static const
drivers/staging/unisys/visorbus/visorchipset.c:1715: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:1829: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visorbus/visorchipset.c:2190: CHECK: Alignment should match open parenthesis
total: 0 errors, 2 warnings, 9 checks, 2458 lines checked
drivers/staging/unisys/visornic/visornic_main.c:276: CHECK: Avoid crashing the kernel - try using WARN_ON & recovery code rather than BUG() or BUG_ON()
drivers/staging/unisys/visornic/visornic_main.c:439: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visornic/visornic_main.c:468: CHECK: Alignment should match open parenthesis
drivers/staging/unisys/visornic/visornic_main.c:1652: CHECK: Alignment should match open parenthesis
total: 0 errors, 0 warnings, 4 checks, 2146 lines checked
drivers/staging/unisys/visorhba/visorhba_main.c:148: ERROR: Macros with complex values should be enclosed in parentheses
total: 1 errors, 0 warnings, 0 checks, 1233 lines checked

selltc, 2016-03-09:
The `WARNING: char * array declaration might be better as static const` problems are now being handled via issue #11.

davidker, 2016-03-09:
The error we have a pass from Dan C. I'm not sure if the pass is good enough to get out of staging, but it is a pass we can argue before we have to do it.

selltc, 2016-03-26:
These should be fixed by my current patches on [my upstream-next-selltc-20160325 branch](/davidker/unisys/commits/upstream-next-selltc-20160325):
commit be8d5071523d84561a547aa789079b25d41a6ffb
commit d849a9a923801b9632d380770294a7a800697a65
commit d1013a128d7863c2c6a833d40cd256a3250d0142
commit a79a3ebbb3a5976683214a62fef52f41de47cd27

selltc, 2016-03-26:
Forgot to say... the previously-mentioned patches passed checkpatch and regression testing.

selltc, 2016-04-12:
Greg accepted all of these except commit a79a3ebbb3a5976683214a62fef52f41de47cd27s (staging: unisys: visornic: convert BUG_ON to dev_err() message), to which he said:

    Please return a real error number, don't make up error numbers like -1

I'll fix and re-submit a corrected patch.

selltc, 2016-04-12:
Corrected patch is commit 09315e9f30936703611218013fc2e061de196cee.

This is compile-clean and checkpatch-clean.

selltc, 2016-06-09:
Greg committed these to staging-next on 4/12:
* commit e4feb2f257955cd9319f38bc935eda99ca6294b6 - staging: unisys: visorbus: 'unsigned' --> 'unsigned int'
* commit a07d7c3858aef317495e061ffc93207c89771fe2 - staging: unisys: visorbus: CHECK: Alignment should match open parenthesis
* commit dc38082f0f0967f8badce0ae5ecf07cd5843d65f - staging: unisys: visornic: CHECK: Alignment should match open parenthesis

Greg committed this to staging-next on 4/29:
* commit 6a957193ba83e5487705f63011307d202fff69f8 - staging: unisys: visornic: convert BUG_ON to dev_err() message

closed 2016-06-09 by selltc