davidker/unisys #12: visordriver_remove_device

davidker, 2016-03-11:
It has a wmb(), why? Shouldn't the lock be good enough? 

Greg yelled at the one in visordriver_probe_device which I'm addressing already.

selltc, 2016-03-11:
Sorry, I should have looked here before mentioning this in my email a few hours ago.  Looks like you already had it covered.  :-)

davidker, 2016-03-14:
I'm not sure why this can't be a lock. The only place dev_start_periodic_work is called is when we call visorbus_enable_channel_interrupts. That should always be callable when we can grab locks so we should be able to just lock the whole block. And if we are being removed while we are in periodic work I think we have problems anyways. I think we need to reinvestigate the locks currently.

selltc, 2016-03-14:
I suspect you are probably correct.  I just knew that locking had changed with your interrupt code, and I didn't want to muddy the waters.

davidker, 2016-03-14:
Understood. I'm debating about either trying to fix it or just getting rid of the wmb(). Can you inspect some patches for me if I send them?

selltc, 2016-03-14:
Sure thing.

davidker, 2016-03-14:
Tim, can you look at 151816a

davidker, 2016-03-31:
in staging-next

closed 2016-03-31 by davidker