davidker/unisys #13: Update comments to match coding standard.

davidker, 2016-03-11:


davidker, 2016-03-11:
I'm not sure if the function comments that we have are up to the coding style of the kernel. Or if it matters, I know that visornic had to have specific function style comments.

gigatwo, 2016-03-15:
According to the CodingStyle document, kernel-doc style comments need to be used for kernel-API functions, it doesn't really say that they need to be there for module functions. But it might not hurt for consistency's sake.

davidker, 2016-04-13:
@selltc Tim, can you look over this set of patches as well?

davidker, 2016-04-13:
upstream-curtinap-kernel-doc

selltc, 2016-04-14:
I added my comments to commit c8d99ccbb3718ffb8d17cc141ca099b7ebb54a82 and commit c93ce4059795840565bed34e80a6f570e506b4ea.  I'll create patches that accommodate these suggestions.

selltc, 2016-04-14:
Tweaked patches accomodating my suggestions: commit cd605f609c6cb276ceb172a1da515e125a65457c, commit 5b2d44813bc280cdfe4bbb1ecbdcda96b08e7c4a, commit 607d696b0e8b847147ee86c9d461b104e38b99c8.
(2 of these patches preserve Alex as the author.)

selltc, 2016-04-14:
BTW, the patches referenced in the preceding comment are checkpatch-clean and compile-clean.

selltc, 2016-04-19:
How's this for a cover letter?:
```
staging: unisys: add polish to structs: visor_driver and visor_device

For 'struct visor_driver' and 'struct visor_device':
* unused members have been removed
* kerneldoc comments have been added
```
Also, all 3 patch comments (commit cd605f6, commit 5b2d448, commit 607d696) need to begin with (several are incorrect):
```
staging: unisys: include:
```

selltc, 2016-04-19:
Note that we can NOT submit this patch-set upstream until the code for issue #19 (commit cad8a4efc1512f0c8bbb88f39a02ebf3f17e6712) is committed to Greg's staging-next branch, which it is NOT yet.  But Greg HAS accepted it...

davidker, 2016-04-19:
Greg updated the patches last night. I had today off, I plan on submitting the patches tomorrow. I'll fix the comments (although technically they only need staging: unisys: The include is just really really nice. I'll add that to my list of accept/rejects.

selltc, 2016-04-19:
I see that the upstream-next versions of this patchset are:
* commit ab202776e648f80844c6dd9f4d2b48704562c50b
* commit 7bd48d351b63a6b7ca5e59b8f0a6ae32a8aed149
* commit 9968639f96587af293901587569eb6778e6d8f75

I was just about to re-write the patch comments for these as-per my suggestions above in  issuecomment-211703410, but then thought better.  Perhaps it is easier to tweak the comments at email generation time.

davidker, 2016-04-19:
I’ve tweaked them in the emails currently. Do you want me to cc on my test send and you sign off on them?

I’m planning on doing the test send in the next 3 minutes.

David Kershner

  *   commit 7bd48d3<https://github.com/davidker/unisys/commit/7bd48d351b63a6b7ca5e59b8f0a6ae32a8aed149>
  *   commit 9968639<https://github.com/davidker/unisys/commit/9968639f96587af293901587569eb6778e6d8f75>

selltc, 2016-04-19:
Sure, if you'd like a second set of eyes...

selltc, 2016-06-09:
Greg committed these to staging-next 4/29:
* commit 6d24b4b10e8a5a299fe0c85060e41a367727efb3 - staging: unisys: include: remove unused fields from struct visor_driver
* commit b9edaf7cebb4ab1e8bef797e89617fa5f113041a - staging: unisys: include: Added kernel-doc for struct visor_driver
* commit 39b92809830b1c0b4d00e934d39a5815c85fd956 - staging: unisys: include: Added kerneldoc comments to visor_device

closed 2016-06-09 by selltc