davidker/unisys #14: Get rid of unused code.

davidker, 2016-03-13:
`static ssize_t chipsetready_store(struct device *dev,
                                  struct device_attribute *attr,
                                  const char *buf, size_t count)
{
        char msgtype[64];

        if (sscanf(buf, "%63s", msgtype) != 1)
                return -EINVAL;

        if (!strcmp(msgtype, "CALLHOMEDISK_MOUNTED")) {
                chipset_events[0] = 1;
                return count;
        } else if (!strcmp(msgtype, "MODULES_LOADED")) {
                chipset_events[1] = 1;
                return count;
        }
        return -EINVAL;
}
`

Do we really need this in a guest?  Isn't CALLHOMEDISK_MOUNTED just a Service Partition call?  Modules_LOADED is that still needed when it gets loaded on demand?

selltc, 2016-03-14:
In my opinion, all this stuff can be removed:
```
chipset_events
check_chipset_events()
clear_chipset_events()
visorchipset_holdchipsetready
chipsetready_store()
visorchipset_guest_attrs
visorchipset_guest_group
```
I also believe that a lot of the code that handles the storing of attributes from usermode is buggy, because it assumes that strings are '\0'-terminated, when in fact they might NOT be, and only `count` bytes are valid.

davidker, 2016-03-14:
Okay, issue it!!

selltc, 2016-03-14:
It seems it might be appropriate to just narrow the scope of this issue to handle the cleaning up of the sysfs attribute stuff, including removing unused code.  So I will rename this issue accordingly.  Feel free to name it otherwise if you disagree.  I figure that we can add other issues to remove other instances of unused code when/if we find them.

davidker, 2016-03-14:
Tim, remember that we have parahotplug user scripts when we clean things up.

davidker, 2016-03-14:
And Tim, I want to keep the sysfs stuff that shows channel attributes because I want to make sure we can get them from the guest environment to compare with the SP. I think that is useful debugging information.

selltc, 2016-03-14:
> Tim, remember that we have parahotplug user scripts when we clean things up.

How do these get distributed with our upstream code?  (Sorry, I'm probably having a brain cramp.)

davidker, 2016-03-14:
the other user app is easy install

selltc, 2016-06-09:
Greg committed this to staging-next on 4/29:
* commit 260d89929c2f45b4b46180e58798bcdee53e0c07 - staging: unisys: visorbus: remove unused chipsetready information

closed 2016-06-09 by selltc