davidker/unisys #19: visorbus_device.responded_to_device_create

gigatwo, 2016-03-23:
This is defined in visorbus.h, but never used. Probably should either remove it figure out why it was there in the first place and do that.

davidker, 2016-03-23:
Thanks Alex, I will look into it. 

David

gigatwo, 2016-03-24:
A fix for this was committed to the issue/19-upstream-next-curtinap-responded_to_device_create branch. 

5eb5c23b6583f1c9cd1083a1a586d7759649610c

gigatwo, 2016-03-24:
@davidker

closed 2016-04-13 by davidker

selltc, 2016-04-14:
I see this being submitted upstream on 4/6 11:20am.  But I don't see Greg ever accepting it.  If that's true, why is this closed?  (I only noticed this because I was working with a patch that wouldn't fit because of this difference between staging-next and upstream-next.)

selltc, 2016-04-14:
This seems to be in Greg's staging-testing, but not staging-next.

davidker, 2016-04-14:
We have gotten emails that it was approved to go to staging-next. I haven't seen him actually update the kernel.org version of staging-next. I should have just moved it to Done but not closed it. Sorry still dealing with KanBoard issues.