davidker/unisys #22: Remove Proc documentation

arfvidsonUIS, 2016-03-24:
https://github.com/davidker/unisys/blob/staging-next/drivers/staging/unisys/Documentation/proc-entries.txt

gigatwo, 2016-03-24:
We should replace it with one mentioning the sysfs entries that we use. :pizza:

```sh
sed -i /proc/sys/g staging/unisys/Documentation/proc-entries.txt
```

TurningWrenches, 2016-04-06:
Approach:
1) Ensure all /proc entries listed in drivers/staging/unisys/Documentation/proc-entries.txt are removed.
2) Remove drivers/staging/unisys/Documentation/proc-entries.txt from repository.
3) Ensure our sysfs entries are documented.

arfvidsonUIS, 2016-04-06:
Don't forget debugfs

TurningWrenches, 2016-04-08:
Steps towards rectification:

1. Ensured all /proc entries listed in /drivers/staging/unisys/Documentation/proc-entries.txt are removed.
2. Deleted /drivers/staging/unisys/Documentation/proc-entries.txt.
3. Ensured the entries in /sys/devices/platform/visorchipset/install, guest, and parahotplug are described in the documentation.
4. Verified the entries in /sys/bus/visorbus/devices/ are described in the documentation.
5. Verified the entries in debugfs do NOT need to be documented at this time.

TurningWrenches, 2016-04-08:
The issue is repaired:

**branch**: proc_documentation
**checkpatch-cleared**: yes, although I received the warning: _WARNING: added, moved or deleted file(s), does MAINTAINERS need updating?_ However, the MAINTAINERS file was updated as appropriate.
**T710-1 verified**: yes

selltc, 2016-06-09:
Greg committed this to staging-next on 4/29:
* commit 538b387234563e030cd9ad8983ad3de8eaf03e13 - staging: unisys: Documentation: Remove proc-entries.txt

closed 2016-06-09 by selltc