davidker/unisys #25: visornic: double-unlock reported by smatch

selltc, 2016-04-05:
See KanBoard 806 - http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=806&project_id=37
```
drivers/staging/unisys/visornic/visornic_main.c:377 visornic_serverdown() error: double unlock 'spin_lock:&devdata->priv_lock'
drivers/staging/unisys/visornic/visornic_main.c:377 visornic_serverdown() error: double unlock 'irqsave:flags'
```
This was originally discussed in #24, but Neil agreed that we can fix the immediate problem involving the double-unlock while in staging, and queue the priv_lock removal (#24) until after we get out of staging.

selltc, 2016-04-07:
Fixed with commit de15885d836d80f73c06da85a973bb751af7aeff.

I unit-tested by adding some messages to visornic_serverdown(), then forcing us thru IO service partition recovery, and ensured I got thru there.  I hit Trac-7234 (https://ustr-svn-1.na.uis.unisys.com/trac/spar/ticket/7234) on the resume path, but that was to be expected.

checkpatch and regression test were clean.

davidker, 2016-04-07:
Tim, I'm having a problem understanding this comment: 

    Previously, devdata->priv_lock was being unlocked in visornic_serverdown()
    both before calling visornic_serverdown(), then again at the end of the
    function.  This bug was corrected.

selltc, 2016-04-07:
Doh! 

_both before calling visornic_serverdown()_
-->
_both before calling visornic_serverdown**_complete()**_

Does that make more sense?

closed 2016-04-13 by davidker

selltc, 2016-05-05:
This is commit 4145ba76b1f7f3296cc673c299084145e1267029 in the upstream.