davidker/unisys #28: remove/fix up TODOs

davidker, 2016-04-14:
Pulled this from the for-later branch (since interrupt todo's have been done). Determine if we need to do them and then write stories, if we don't need them get rid of the TODOs.

kershnda@USTR-ERL-4458[]:~/kernel/master/unisys/drivers/staging/unisys$ find . -print | xargs grep -r -n TODO
./visorbus/visorchipset.c:435:  PARSERSTRING_NAME, /* TODO: only PARSERSTRING_NAME is used ? */
./TODO:1:TODO:
./visorhba/visorhba_main.c:604: *       TODO: Make this per vhba
./include/guestlinuxdebug.h:138:/* TODO-> Info currently doesn't show, so we set info=warning */
./visorbus/visorchipset.c:435:  PARSERSTRING_NAME, /* TODO: only PARSERSTRING_NAME is used ? */
./visorbus/visorchipset.c:435:  PARSERSTRING_NAME, /* TODO: only PARSERSTRING_NAME is used ? */
./TODO:1:TODO:
./visorhba/visorhba_main.c:604: *       TODO: Make this per vhba
./visorhba/visorhba_main.c:604: *       TODO: Make this per vhba
./include/guestlinuxdebug.h:138:/* TODO-> Info currently doesn't show, so we set info=warning */
./include/guestlinuxdebug.h:138:/* TODO-> Info currently doesn't show, so we set info=warning */
kershnda@USTR-ERL-4458[]:~/kernel/master/unisys/drivers/staging/unisys$

davidker, 2017-01-10:
This is closed in for-later. @selltc should we leave this open until we are in staging-next?

selltc, 2017-01-10:
I would think close it.

closed 2017-01-10 by davidker