davidker/unisys #29: Sonarqube visorbus_main.c unused functions

arfvidsonUIS, 2016-04-18:
The function '__check_debugref' is never used
The function '__check_forcenomatch' is never used
The function '__check_forcematch' is never used
The function '__check_debug' is never used
struct or union member 'channel_size_info::max_size' is never used.
struct or union member 'channel_size_info::min_size' is never used

selltc, 2016-05-16:
Dave Binder's patch (sent via email):
```
--- a/drivers/staging/unisys/visorbus/visorbus_main.c
+++ b/drivers/staging/unisys/visorbus/visorbus_main.c
@@ -1260,8 +1260,6 @@ chipset_device_resume(struct visor_device *dev_info)
 
 struct channel_size_info {
 	uuid_le guid;
-	unsigned long min_size;
-	unsigned long max_size;
 };
 
 int
@@ -1329,16 +1327,3 @@ visorbus_exit(void)
 	remove_bus_type();
 }
 
-module_param_named(debug, visorbus_debug, int, S_IRUGO);
-MODULE_PARM_DESC(visorbus_debug, "1 to debug");
-
-module_param_named(forcematch, visorbus_forcematch, int, S_IRUGO);
-MODULE_PARM_DESC(visorbus_forcematch,
-		 "1 to force a successful dev <--> drv match");
-
-module_param_named(forcenomatch, visorbus_forcenomatch, int, S_IRUGO);
-MODULE_PARM_DESC(visorbus_forcenomatch,
-		 "1 to force an UNsuccessful dev <--> drv match");
-
-module_param_named(debugref, visorbus_debugref, int, S_IRUGO);
-MODULE_PARM_DESC(visorbus_debugref, "1 to debug reference counting");
```
`forcematch` and `forcenomatch` are still used (search for `visorbus_forcematch` and `visorbus_forcenomatch`), but you can delete the other 2.  It looks like you can also delete the entire definition of `struct channel_size_info`, because I don't see that being used anywhere.  You also need to delete the corresponding module parameter variables themselves, `visorbus_debug` and `visorbus_debugref`.  I.e.
```
@@ -27,10 +27,9 @@
 #define MYDRVNAME "visorbus"

 /* module parameters */
-static int visorbus_debug;
 static int visorbus_forcematch;
 static int visorbus_forcenomatch;
-static int visorbus_debugref;
+
 #define SERIALLOOPBACKCHANADDR (100 * 1024 * 1024)

 /* Display string that is guaranteed to be no longer the 99 characters*/
@@ -1258,12 +1257,6 @@ chipset_device_resume(struct visor_device *dev_info)
        initiate_chipset_device_pause_resume(dev_info, false);
 }

-struct channel_size_info {
-       uuid_le guid;
-       unsigned long min_size;
-       unsigned long max_size;
-};
-
 int
 visorbus_init(void)
 {
@@ -1329,9 +1322,6 @@ visorbus_exit(void)
        remove_bus_type();
 }

-module_param_named(debug, visorbus_debug, int, S_IRUGO);
-MODULE_PARM_DESC(visorbus_debug, "1 to debug");
-
 module_param_named(forcematch, visorbus_forcematch, int, S_IRUGO);
 MODULE_PARM_DESC(visorbus_forcematch,
                 "1 to force a successful dev <--> drv match");
@@ -1340,5 +1330,3 @@ module_param_named(forcenomatch, visorbus_forcenomatch, int, S_IRUGO);
 MODULE_PARM_DESC(visorbus_forcenomatch,
                 "1 to force an UNsuccessful dev <--> drv match");

-module_param_named(debugref, visorbus_debugref, int, S_IRUGO);
-MODULE_PARM_DESC(visorbus_debugref, "1 to debug reference counting");
```

davidker, 2016-05-16:
Might want to do a series with separate patches ;) Just a thought.

TurningWrenches, 2016-05-16:
**branch:** visorbus_remove_unused_func, series: 23247e0, 302fb8b
**checkpatch-cleared:** yes
**T710-1 verified:** (pending)

selltc, 2016-05-16:
`Reviewed-by: Tim Sell <Timothy.Sell@unisys.com>`

selltc, 2016-08-22:
8/15: Greg committed these to staging-next in:
* commit f4211d1a6b3c77dced986aea4eb0023f1d6dc6c1- staging: unisys: visorbus: remove unused module parameters
* commit cc9b8ed99e4b6e48ae9b28b295ce08fe18bc949c - staging: unisys: visorbus: remove unused struct

closed 2016-08-22 by selltc