davidker/unisys #30: Sonarqube visorchipset unused functions and struct members

arfvidsonUIS, 2016-04-18:
struct or union member 'putfile_active_buffer::pnext' is never used.
struct or union member 'putfile_request::file_request_number' is never used.
struct or union member 'putfile_request::data_sequence_number' is never used. 
The function '__check_major' is never used.
The function '__check_visorbusregwait' is never used.
The function '__check_visorbusregwait' is never used.

arfvidsonUIS, 2016-04-20:
they are used in module_param_named(major, visorchipset_major, int, S_IRUGO)
and
module_param_named(visorbusregwait, visorchipset_visorbusregwait, int, S_IRUGO)

So they maybe needed. So I'm only going to remove  the unused struct members until someone reviews it.

davidker, 2016-05-06:
Made comments based on the patch. I think we need to get rid of one more comment.

selltc, 2016-05-06:
Shouldn't we be linking to a patch from here?  Could someone supply a link to the commit?

closed 2016-05-18 by arfvidsonUIS

selltc, 2016-06-09:
Greg committed this to staging-testing 5/6, and later to staging-next on 5/9:
* commit 456743c357895678431c91c7305d8abaabc92781 - staging: unisys: remove unused struct members