davidker/unisys #31: SonarQube visorbus_main.c %Lx in format string (no. 1) requires 'unsigned long long' but the argument type is 'unsigned long long'

arfvidsonUIS, 2016-04-18:
%Lx in format string (no. 1) requires 'unsigned long long' but the argument type is 'unsigned long long'
All the lines that have this issue:
255
277
369
394
403

davidker, 2016-04-19:
Can we not ignore this one. I think it comes from SonarQube not recognizing that u64 is just another name for unsigned long long. I think the fact it doesn't check the two string values to be identical is a bug in the program. I wonder if we can fix that and resubmit it back to the community.

arfvidsonUIS, 2016-04-20:
Just push branch in case we need it gihubissue#31

closed 2016-05-18 by arfvidsonUIS

selltc, 2016-06-09:
Greg committed this to staging-testing 5/6, and later to staging-next 5/9:
* commit 8a1cca31abd4d6470b8fe6f1b2a6fc2f8868ca8b - staging: unisys: fix format string %Lx to %llx for u64