davidker/unisys #38: visorhba: process_disk_notify(): switch success handling to error handling

selltc, 2016-05-02:
Source: Dan Carpenter dan.carpenter@oracle.com Mon 5/2/2016 5:53 AM
> process_disk_notify()
> switch success handling to error handling.

selltc, 2016-05-05:
@TurningWrenches, it looks like your fix is commit d029a8c48c89bb7726f2197839063c0d46eaa856.  I have NO IDEA why I got an email about that patch, but it looks fine to me.

TurningWrenches, 2016-05-05:
That's funny, because I didn't receive an e-mail that you responded. I just happened to look at this issue and saw your response :)

davidker, 2016-05-05:
Where is the fix for this? @TurningWrenches

davidker, 2016-05-05:
New commit ID was found at 1b54049435869cafe6b6d274da2377534f9072cf

TurningWrenches, 2016-05-08:
branch: process_disk_notify_err_handling, commit 1b54049
checkpatch-cleared: yes
T710-1 verified: yes

selltc, 2016-06-09:
5/9 - Greg committed this to staging-testing and staging-next:
* commit 3b5e11d7f566495f7ff3f4178b814a59b4ea84a5 - staging: unisys: visorhba: switch success handling to error handling

closed 2016-06-09 by selltc