davidker/unisys #43: visornic: visornic_set_multi(): simplify if statement

selltc, 2016-05-02:
```
1019   */
1020  static void
1021  visornic_set_multi(struct net_device *netdev)
1022  {
1023          struct uiscmdrsp *cmdrsp;
1024          struct visornic_devdata *devdata = netdev_priv(netdev);
1025  
1026          /* any filtering changes */
1027          if (devdata->old_flags != netdev->flags) {
1028                  if ((netdev->flags & IFF_PROMISC) !=
1029                      (devdata->old_flags & IFF_PROMISC)) {
```
Source: Dan Carpenter dan.carpenter@oracle.com Mon 5/2/2016 5:53 AM
> Reverse these if statements and pull the code in two indent levels.

TurningWrenches, 2016-05-05:
**branch:** process_disk_notify_err_handling
**checkpatch-cleared:** yes
**T710-1 verified:** yes

See commit: 7f43241

selltc, 2016-05-06:
This patch looks good to me.  Feel free to mark:
```
Reviewed-by: Tim Sell <Timothy.Sell@unisys.com>
```

selltc, 2016-06-09:
6/8 - Greg committed this to staging-testing and staging-next:
* commit 6d8c96cbc161168726480ab01ad5c53fd6b2e04a - staging: unisys: visornic: simplify visornic if statements

closed 2016-06-09 by selltc