davidker/unisys #44: visornic: visornic_rx(): clean up issues involving return value

selltc, 2016-05-02:
```
1154  /**
1155   *      visornic_rx - Handle receive packets coming back from IO Part
1156   *      @cmdrsp: Receive packet returned from IO Part
1157   *
1158   *      Got a receive packet back from the IO Part, handle it and send
1159   *      it up the stack.
1160   *      Returns void
```
Source: Dan Carpenter dan.carpenter@oracle.com Mon 5/2/2016 5:53 AM
> Actually it returns either zero or one but in a convoluted way.  Update
> it to use literals and update the comment.

davidker, 2016-05-09:
Patch?

TurningWrenches, 2016-05-12:
branch: visornic_rx_rtn_val, commit 6422c33
checkpatch-cleared: 
T710-1 verified:

selltc, 2016-06-09:
6/8 - Greg committed this to staging-testing and staging-next:
* commit 73e81350ad7a45117c17fb3ec433a266b3fa7d78 - staging: unisys: visornic: change return statements

closed 2016-06-09 by selltc