davidker/unisys #45: visornic: devdata_initialize(): remove extraneous error-check

selltc, 2016-05-02:
```
1361  /**
1362   *      devdata_initialize      - Initialize devdata structure
1363   *      @devdata: visornic_devdata structure to initialize
1364   *      #dev: visorbus_deviced it belongs to
1365   *
1366   *      Setup initial values for the visornic based on channel and default
1367   *      values.
1368   *      Returns a pointer to the devdata if successful, else NULL
1369   */
1370  static struct visornic_devdata *
1371  devdata_initialize(struct visornic_devdata *devdata, struct visor_device *dev)
1372  {
1373          if (!devdata)
1374                  return NULL;
```
Source: Dan Carpenter dan.carpenter@oracle.com Mon 5/2/2016 5:53 AM
> This check doesn't make sense and the error handling is kind of odd.
> Just remove it, remove the error handling and update the comment.

TurningWrenches, 2016-05-09:
**branch:** visornic_remove_check, commit fb28248
**checkpatch-cleared:** yes
**T710-1 verified:** (pending)

selltc, 2016-05-09:
Thanks David.
```
Reviewed-by: Tim Sell <Timothy.Sell@unisys.com>
```

selltc, 2016-06-09:
6/8 - Greg committed this to staging-testing and staging-next:
* commit e1834bd0f6e2e00b04dc992d15051fb43790814e - staging: unisys: visornic: remove extraneous error check

closed 2016-06-09 by selltc