davidker/unisys #46: visornic:visornic_probe(): cleanup error-handling attempts to free things never allocated

selltc, 2016-05-02:
Dan's complaint is specifically about rcvbuf, although it appears to me that we have the same issue for many of our other error-exit cases.
```
1817          err = visorbus_read_channel(dev, channel_offset,
1818                                      &devdata->num_rcv_bufs, 4);
1819          if (err) {
1820                  dev_err(&dev->device,
1821                          "%s failed to get #rcv bufs from chan (%d)\n",
1822                          __func__, err);
1823                  goto cleanup_netdev;
1824          }
1825  
1826          devdata->rcvbuf = kcalloc(devdata->num_rcv_bufs,
1827                                    sizeof(struct sk_buff *), GFP_KERNEL);
1828          if (!devdata->rcvbuf) {
1829                  err = -ENOMEM;
1830                  goto cleanup_rcvbuf;
```
Source: Dan Carpenter dan.carpenter@oracle.com Mon 5/2/2016 5:53 AM
> We never allocated rcvbuf.  It should still be goto cleanup_netdev;  The
> other gotos after this in this function are slightly off as well.  But
> thanks for the nice label names because it makes reviewing this much
> easier.  :)

TurningWrenches, 2016-05-08:
**branch:** visornic_cleanup_err_handling, commit b98785b
**checkpatch-cleared:** yes
**T710-1 verified:** pending

selltc, 2016-06-09:
6/8 - Greg committed this to staging-testing and staging-next:
* commit d12324e37d4f384b93df65d5188a1875ee80ed4c - staging: unisys: visornic: cleanup error handling

closed 2016-06-09 by selltc