davidker/unisys #47: visornic: visornic_init(): switch error-handling and success-handling

selltc, 2016-05-02:
```
2126          err = visorbus_register_visor_driver(&visornic_driver);
2127          if (!err)
2128                  return 0;
```
Source: Dan Carpenter dan.carpenter@oracle.com Mon 5/2/2016 5:53 AM
> Flip this around.
```
if (err)
    goto cleanup_debugfs;
 
return 0;
```

TurningWrenches, 2016-05-09:
**branch:** visornic_switch_handling, commit 51f6a1e
**checkpatch-cleared:** yes
**T710-1 verified:** (pending)

selltc, 2016-05-09:
```
Inspected-by: Tim Sell <Timothy.Sell@unisys.com>
```

selltc, 2016-06-09:
6/8 - Greg committed this to staging-testing and staging-next:
* commit 186896fdf0fd7d4617435986ed79f817eecd3c0f - staging: unisys: visornic: check for error instead of success

closed 2016-06-09 by selltc