davidker/unisys #54: 'for-later' interrupts do NOT work in SLES12.0 or RHEL7.2

selltc, 2016-05-18:
[KanBoard-979](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=979&project_id=37)

selltc, 2016-05-18:
Fixed by commit 57781f99da5a8cb7ae3c1898c9dfb42f7f87111b.

selltc, 2016-05-18:
If I had to take a guess, I would say that it is commit 527c0b19700536f86c758c12f00b9d48bc7ee194 that is at fault, because that is where `channel_interrupt` **SHOULD** have first been assigned a value of visorhba_isr(), but was **NOT**.  I.e., I think this is a mistake where a necessary change was omitted, rather than a change that messed it up.

selltc, 2016-05-18:
I went ahead and commited this to our for-later svn mirror in [r29725](https://ustr-svn-1.na.uis.unisys.com/trac/spar/changeset/29725).

davidker, 2016-05-18:
Thanks! I've incorporated it into for-later which should be up later today.

selltc, 2016-05-19:
Thanks David.  I now see this fixed in your revised for-later commit 42e3e0a8394e481c0991639bbaac12beee2901f2.

selltc, 2016-09-06:
In today's version of for-later, verified this is in commit 370f552ac18b871470d55295e4065c556ffc559d.

davidker, 2017-01-10:
@selltc, can we close this? Or do we keep this open to do checks?

michael-didomenico, 2017-01-10:
I can't figure why we didn't close this a long time ago...

selltc, 2017-01-10:
(Sorry... I was accidentally logged in to Mike's account before, as I need that to access SDG's github...)
I can't figure why we didn't close this a long time ago...

davidker, 2017-01-10:
Closing.. okay I was really confused there for a second.

closed 2017-01-10 by davidker