davidker/unisys #55: visorbus: visordriver_callback_lock should be a mutex, not a semaphore

selltc, 2016-05-19:
Source: Thomas Gleixner <tglx@linutronix.de> 5/18/2016

mriswyth, 2016-05-20:
visordriver_callback_lock converted to a mutex in githubissue#55-upstream-next.

5cd97a915b5a32142902f06a60f915128782582e

selltc, 2016-06-01:
Comments on this patch were provided by Thomas Gleixner tglx@linutronix.de 6/1
> ```
> --- a/drivers/staging/unisys/include/visorbus.h
> +++ b/drivers/staging/unisys/include/visorbus.h
> @@ -161,7 +161,8 @@ struct visor_device {
>  	struct timer_list timer;
>  	bool timer_active;
>  	bool being_removed;
> -	struct semaphore visordriver_callback_lock;
> +	/* mutex to serialize visor_driver function callbacks */
> ```
> 
> TBH. I hate these kind of comments. The mutex name is self explaining, right?
> I rather wish you would have spent time documenting the non obvious parts of
> the code.

selltc, 2016-08-22:
8/15: Greg applied this to staging-next with:
* commit d505855e5f2022abc75d1a54bafeb80f25a955a7 - staging: unisys: visorbus: Make visordriver_callback_lock a mutex

closed 2016-08-22 by selltc