davidker/unisys #60: visorbus: gripes about kerneldoc function comments

selltc, 2016-05-19:
Source: Thomas Gleixner tglx@linutronix.de 5/18/2016:
> Try to mimic kerneldoc comments, i.e. start with: /**
> but do not implement any of the kerneldoc requirements.

TurningWrenches, 2016-05-20:
**branch:** adjust_visorbus_func_comments: commit 4b5a11d642608b1f7051aba6cabe850d399212ec
3028fe3bf6e09736fbe715a6514283657ec58ee0
62cfa1195fbcf61dacc3308de1efd1352dc35ebb
72b5f18f2e919cf8ff25a115e058426c264ff0be
3443add77a388868c3f903840068a88b9ccbe67a
**checkpatch-cleared:** yes
**T710-1 verified:** (pending)

selltc, 2016-05-22:
I could be wrong, but it is my belief that what tglx was complaining about was the lack of standard structured kerneldoc comments.  Have a peek at Documentation/kernel-doc-nano-HOWTO.txt in the kernel source tree, which has an example like this:
```
**
 * foobar() - short function description of foobar
 * @arg1:   Describe the first argument to foobar.
 * @arg2:   Describe the second argument to foobar.
 *      One can provide multiple line descriptions
 *      for arguments.
 *
 * A longer description, with more discussion of the function foobar()
 * that might be useful to those using or modifying it.  Begins with
 * empty comment line, and may include additional embedded empty
 * comment lines.
 *
 * The longer description can have multiple paragraphs.
 *
 * Return: Describe the return value of foobar.
 */

```
In other words, tglx wants to see:
* the "short function description" on the first line
* a description of each argument as "`@<argname1>`", "`@<argname2>`", etc.
* a longer description of the function
* a description of the return value

But I could be wrong; I would be interested to hear the interpretation of others.  Unfortunately, my interpretation would require a LOT more work.  :-(

selltc, 2016-05-22:
I should clarify what I said a bit.  I do **NOT** think it is necessary that we provide kerneldoc comments for all functions, as indicated by this comment in Documentation/kernel-doc-nano-HOWTO.txt:
```
We definitely need kernel-doc formatted documentation for functions
that are exported to loadable modules using EXPORT_SYMBOL.

We also look to provide kernel-doc formatted documentation for
functions externally visible to other kernel files (not marked
"static").

We also recommend providing kernel-doc formatted documentation
for private (file "static") routines, for consistency of kernel
source code layout.  But this is lower priority and at the
discretion of the MAINTAINER of that kernel source file.
```

However, for functions that we deem important enough to document (as-per the above guidelines), we should always use kerneldoc comments using the recommended conventions.

TurningWrenches, 2016-05-23:
I ultimately interpreted the feedback to ask for kerneldoc-style wrapping (i.e. begin with `/**` on its own line, and end with `*/ on its own line), but to leave out the kerneldoc style formatting within. But I also could be wrong, and would be interested in hearing other opinions.

For what its worth, I did begin implementing a full kerneldoc workup last week, should what I committed not be enough.

[kerneldoc.zip](https://github.com/davidker/unisys/files/277611/kerneldoc.zip)

TurningWrenches, 2016-05-26:
More commits:
**branch:** visorbus_kernel_doc
1013c601e6cd9024e7ecd2200dcc5348fb35a906
6778a0739cd8735bb214772034a016a41cc60a82
**checkpatch-cleared:** yes
**T710-1 verified:** (pending)

selltc, 2016-06-01:
Comments on this from Thomas Gleixner <tglx@linutronix.de> 6/1:
> That was not my feedback. If you start a comment with `'/**'` then it should be
> a kernel-doc function/struct documentation. If not then it starts with `'/*'`.

TurningWrenches, 2016-06-01:
Per my comments above, 7 commits were made to the <code>visorbus_kernel_doc</code> branch. Of these, I verified that all 7 were committed to the upstream, and therefore need to be reconsidered:
 4b5a11d642608b1f7051aba6cabe850d399212ec
3028fe3bf6e09736fbe715a6514283657ec58ee0
62cfa1195fbcf61dacc3308de1efd1352dc35ebb
72b5f18f2e919cf8ff25a115e058426c264ff0be
3443add77a388868c3f903840068a88b9ccbe67a
6778a0739cd8735bb214772034a016a41cc60a82
1013c601e6cd9024e7ecd2200dcc5348fb35a906

TurningWrenches, 2016-06-01:
My corrections, per Thomas Gleixner's comments, can be found here:

**branch:**
**checkpatch-cleared:**
**T710-1 verified:**

TurningWrenches, 2016-06-01:
This commit adds the appropriate @ prefix before the members of struct visor_device:

a51545f1176e45ceff3f52a988ef27f7a101bed8

selltc, 2016-06-03:
For our latest-round of enhancements to go into our upcoming v3 patches:
* visorbus_main.c changes are in commit 5eb5a3063e19eeb546459c5384728a15b7360000.  For functions that Dave B already documented in visorbus.h, that patch just copies Dave's text into visorbus_main.c.  For all other functions, I added kerneldoc.  That patch at least compiles, and does NOT introduce checkpatch errors.
* I also have commit 5dab5be94f1d1077845c5062722d7da6884df4b5, which removes all of the kerneldoc comments, but that also removes comments that should be moved into visorchannel.c, which I did not do.

selltc, 2016-06-03:
I updated the visorbus_main.c changes, and pushed a new branch.  I updated the previous comment so that it references the new corrected commits.

TurningWrenches, 2016-06-03:
Comment changes for visorchipset.c (v3 submission): df0cfeb1cba5d14ed03e4a41fb4b87e2e01735ea

selltc, 2016-06-03:
I think these are the last commits we need to complete this then:
* commit 59affcdb041723b22864a756e1c79ce6e0948601 - staging: unisys: visorbus: remove unused parameter from function (because I couldn't hardly document a parameter that was NOT even used)
* commit 506b215f606ae2b6fd0cbcc0644dd113fa19180e - staging: unisys: visorbus: visorchannel.c, visorbus_private.h comments
* commit 2131c1dcd59768746bdf44c07fcced5bc7733859 - staging: unisys: visorbus: fix commenting in vbusdevinfo.h

selltc, 2016-06-04:
As-per David's request, I split commit 506b215f606ae2b6fd0cbcc0644dd113fa19180e into commit 6d82f9ab13a52633e929f9bd16a14e2348b5d821 and commit 0e12688156efe4c437fb787fedd2e78599d6b32b, in preparation for submitting v3 of our visorbus patchset.

selltc, 2016-08-22:
8/15 - Greg committed these patches into staging-next with:
* commit 93e59bfc3bd0b4069b06cd42025e6735b9aecf5b - staging: unisys: visorbus: remove unused parameter from function
* commit 0048be9e1ee99d6c63001aaf831d797c3fa8e345 - staging: unisys: visorbus: fix commenting in vbusdevinfo.h
* commit 3fd1b3b6829c5ccbdc5af9a86e59d6360af9972b - staging: unisys: visorbus: fix commenting in visorbus_main.c
* commit e19674ceb6f2b07d4c4433e0b805616f58b2983c - staging: unisys: visorbus: fix visorchannel.c comments
* commit ec17f452f4e9eb279d5dd52966b99c055a450028 - staging: unisys: visorbus: Rectify commenting in visorchipset.c
* commit e9b18f3b865e32f927e9597a270e684a6984f0e5 - staging: unisys: visorbus: Move visorbus-unique functions to private header
* commit eafc6a94e93f642ba0e15fc4593b008c67c0cd10 - staging: unisys: visorbus: rectify kerneldoc comment for struct
* commit 90bb5c1f27d2af7e0a8430c58fa83f8d910a9e44 - staging: unisys: visorbus: fix visorbus_private.h comments

closed 2016-08-22 by selltc