davidker/unisys #63: visorbushelper.h has include dependencies

mriswyth, 2016-05-23:
During the move of visorbus out of staging, it was discovered that visorbushelper.h had include issues when I changed the order of the includes in visorbus. That caused me to notice that it was only used in the visorbus driver, so I moved it over to the visorbus directory. It would be nice however to determine what the dependencies are with the other include files and clean them up.

[Kanban-1212](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=1212&project_id=37)

mriswyth, 2016-05-23:
Looks like visorbushelper.h might actually be vbushelper.h.

mriswyth, 2016-05-25:
The problem was that vbushelper.h should have included vbusdeviceinfo.h so that it contained the definition for ultra_vbus_deviceinfo.

However, since vbushelper.h is only needed by visorbus the fix I made was to just include the few contents of vbushelper.h in visorbus_private.h.

* a2de03e35c9a738c7653b0bb3263f68539bd5494 - Changeset causes 1 checkpatch warning that I'm deleting a file and MAINTAINERS may need updated. I don't think MAINTAINERS needs updated.

Branch with fix is githubissue#63-upstream-next.

selltc, 2016-09-13:
Greg committed this to staging-next on 9/12 as:
* commit 43c0ab0e1c7f9d5dec502af988b8922599b485f5 staging: unisys: Move vbushelper.h contents to visorbus_private.h

So I am closing this issue.

closed 2016-09-13 by selltc