davidker/unisys #66: ultrainputreport.h includes itself, REALLY?

mriswyth, 2016-05-25:
Noticed when looking through include files that ultrainputreport.h includes itself??!!?

```
#ifndef __SPAR_ULTRAINPUTREPORT_H__
#define __SPAR_ULTRAINPUTREPORT_H__

#include <linux/types.h>

#include "ultrainputreport.h"
```
[KanBan-1289](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=1289&project_id=37)

mriswyth, 2016-05-25:
Removed ultrainputreport.h from ultrainputreport.h.
Also removed channel.h from visorinput.c since it was unnecessary.

* ce2f5e54dabc4a5c84551c190f773e8737b5689e

[githubissue#66-upstream-next](https://github.com/davidker/unisys/tree/githubissue%2366-upstream-next)

davidker, 2016-06-09:
@mriswyth , can you split this out into two patches? Tim Sell just got pinged for having an also in his comment meant it should be two patches. Since it was just comments he let it slide.

mriswyth, 2016-06-09:
@davidker Split into two patches in [githubissue#66-upstream-next-2](https://github.com/davidker/unisys/tree/githubissue%2366-upstream-next-2).

Commits
* 2f0b427237c05d84e8a99c3468c890f68259b200
* bd1f2e804476f804262d593effbd426beb204853

I'm happy for you to choose which ever version of the patch/patches you want to push, so I'll point out that the original patch (ce2f5e54dabc4a5c84551c190f773e8737b5689e) was worded specifically to make it look like I was doing 1 thing in the patch (removing #included files). There is no also or and in the patch comment, that just popped up in my github issue comment.

selltc, 2016-09-13:
Greg applied the above-mentioned commits to staging-next on 9/12, as:
* commit 89cb05105117dfa0b86635c7e8c124e529047f15 Bryan Thompson staging: unisys: visorinput: Remove ultrainputreport.h include
* commit d7cbd9f6c4cdb379ebd04cda9cfce1bbf6442e26 Bryan Thompson staging: unisys: visorinput: Update visorinput.c includes

So I think this can be closed now.

closed 2016-09-13 by selltc