davidker/unisys #67: visorbus_private.h called visorchipset.h internally

mriswyth, 2016-05-25:
visorbus_private.h is labelled as visorchipset.h at the top of the header. Its include protection wrapper is also __VISORCHIPSET_H__.

There are also places where visorchipset is referred to that may need to be changed to visorbus or made explicitly visorchipset.c.

[KanBan-1340](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=1340&project_id=37)

TurningWrenches, 2016-06-06:
**Branch:** visorbus_fix_header_names; commits: adccee33c456716ccbb98b789e2e0d1458f705be
**checkpatch-cleared:** yes
**T710-1 verified:** no

selltc, 2016-06-06:
Other than a typo in the commit e7cc3264cd79fd648efe10c29f22fab3b5295cbc log comment: "longe" --> "longer", this looks fine.

You might also shorten the commit adccee33c456716ccbb98b789e2e0d1458f705be headline to 72 characters, but I'm not sure that's a submission requirement.  (I know github is more friendly with 72 characters for some reason.)

TurningWrenches, 2016-06-06:
Thanks, Tim! I've implemented the changes you suggested, and updated the SHA1s.

selltc, 2016-06-06:
Thanks David; looks good.

selltc, 2016-09-13:
Greg committed one of these patches to staging-next on 9/12 as:
* commit 7339cb09a91ea316b6b84944910a2869abfc7fea staging: unisys: visorbus: Convert visorchipset.h refs

But I do NOT see the "staging: unisys: visorbus: Convert references to /proc to /sys" patch committed anywhere yet, so I'm leaving this issue open for now.

davidker, 2016-12-07:
Sorry I missed 7d0ce58. I will try to figure out what happened and submit it upstream.

davidker, 2017-01-10:
It was committed as patch 5d501ef

closed 2017-01-10 by davidker