davidker/unisys #69: visorbus: Remove visorbus registration with visorchipset

TurningWrenches, 2016-05-26:
Visorchipset and visorbus are the same driver now so no need to wait around for them to be registered not registered. Need to clean it up and simplify it.

TurningWrenches, 2016-05-26:
**Branch:** remove_visorbus_registration_w_visorchipset

selltc, 2016-05-26:
@TurningWrenches, re commit 34ccf2e6162caff5042b41df32e6f49362273d67, I think you perhaps took @davidker too literally here.  ;-)

Presently, visorbus requires this registration in order to operate correctly, because it supplies the necessary callback functions to create busses, devices, etc., as messages are received via the controlvm channel.  (I can't see how anything could possibly work by just stripping it out.)  This made a lot of sense when visorchipset and visorbus were different drivers, but just adds complication now that they have been combined.

I think @davidker intended for us to re-structure the driver so that this registration was no longer necessary.  That may be a moderately-complicated undertaking.

davidker, 2016-05-26:
Correct Tim, and David B knows it. These are introduction patches to understanding the full effort. None of this is going in anytime soon.

selltc, 2016-05-26:
Oops... <blush> sorry... I jumped the gun.

selltc, 2016-06-01:
Thomas Gleixner <tglx@linutronix.de> pointed out during his comments on 6/1 about the need to "convert notifier_lock to mutex", but my understanding about what I just heard from @TurningWrenches and @davidker is that this set of patches will be removing this lock.

selltc, 2016-06-01:
I can't convey to you how much I LOVE this patchset!  (grokking this code will be so much more pleasant now)

The only thing I would change is these sentences in the patch comments for the first 2 patches in the set:
```
Previously, visorbus and visorchipset were combined into one driver. As a
result, there is no need to register visorbus with visorchipset any longer.
```

I would clarify it like this:
```
When this functionality was first implemented, visorchipset and visorbus were
separate drivers, which necessitated a registration mechanism for them to
communicate.  More-recently, visorchipset and visorbus were combined into a
single driver, and now exist as separate source files within the same driver,
known as 'visorbus'.  This eliminated the need for a registration mechanism,
but it has remained nevertheless until now.  For the sake of simplification,
this registration mechanism is now being removed.
```

TurningWrenches, 2016-06-01:
Thanks, Tim! :)

I like the clarification a lot. I will put it in, and then re-push.

TurningWrenches, 2016-06-02:
Ok, I updated the commit comments, and refreshed the SHA1s above. Thanks!

davidker, 2016-06-03:
Okay, I might be missing something, but if you do the notifiers first and get rid of the registration function in that patch aren't you messing up the registration of the responders as well? Aren't we going to be calling NULL devices or not responding and leaving the host in a Starting state? I'm running tests, but it would seem to NOT be a good idea to do it in this order. I'm tried to reverse the patches mechanically but that didn't work, I might try it a little more surgical next time.

selltc, 2016-06-03:
VERY good point, David.  Sorry I missed that.  Could just squashing the patches be the answer?

davidker, 2016-06-03:
Ideally I think they should be reversed but squashing the two patches might be the best for the short term and be good enough. 

I'll squash/revise comments and send up.

selltc, 2016-08-22:
8/15: Greg applied these patches to staging-next with:
* commit 87241ab85930519a962a567de58d959d3029b419 - staging: unisys: visorbus: Remove notifier-related code from visorbus
* commit ea3a5aafe0fbd2d10090ea9f0695d8700f89190b - staging: unisys: visorbus: Rename function to follow existing convention

closed 2016-08-22 by selltc