davidker/unisys #75: visorhba interrupt_rcvd read in wait but it is never set

mriswyth, 2016-06-14:
visorhba's process_incoming_rsps contains a [wait_event_interruptible_timeout](https://www.kernel.org/doc/htmldocs/device-drivers/API-wait-event-interruptible-timeout.html) call that is waiting for interrupt_rcvd to be equal to 1, but interrupt_rcvd is never set.

```
                wait_event_interruptible_timeout(
                        devdata->rsp_queue, (atomic_read(
                                             &devdata->interrupt_rcvd) == 1),
                                msecs_to_jiffies(devdata->thread_wait_ms));
```

This smells odd, do we think it is right? If it is then perhaps a comment is in order.

mriswyth, 2016-06-14:
[Kanban 1587](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=1587&project_id=37)

selltc, 2016-06-14:
Unless I'm missing something, this code doesn't exist in for-later, so I'm not sure if it's worth messing with.

davidker, 2016-06-14:
Yeah.. but now I'm curious about this. 

Maybe that is why we have major performance issues.

davidker, 2016-06-14:
Okay, so it is just forcing a poll wait before looking at the queue and a similar issue with interrupt_rcvd is happening in the visornic driver as well. Wonder if anyone will notice.

selltc, 2016-06-14:
Yeah, we will still timeout every 2ms (devdata->thread_wait_ms), so we'll poll 500 times/second.

mriswyth, 2016-06-15:
Since this wait is necessary now and is removed in for-later there isn't a benefit to doing anything with the current version. I noticed that for-later still contains the now unreferenced interrupt_rcvd so I'm going to create a Kanban item to remove interrupt_rcvd from for-later since for-later isn't in a state to have patches put in right now.

closed 2016-06-15 by mriswyth