davidker/unisys #82: Task #24495 - Get rid of unused variables

jaftrain, 2016-08-24:
[http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=24495&project_id=37](Kanboard-24495)

selltc, 2016-09-13:
Greg committed the 3 commits referenced above to staging next on 9/12:
* commit 1e5bcbe67aee51e72e3a0411351d783185c8229b staging: unisys: visorbus: remove putfile_buffer_entry
* commit d5bc757c1f6606c03e14d093c3a96e9e7f2f413f staging: unisys: visorbus: remove unused lists
* commit 1bda5a2b0acc4f99b602f612c24da777a989c095 staging: unisys: visorbus: remove g_devicechangestate_packet

If there is no more work remaining on this issue, please close it.

jaftrain, 2016-09-15:
There are two more patches in this branch that are being sent up to Greg in next set. Once (if) Greg commits them I will close the issue. 
commit 5c833b3c53aebf61a4ddd295f37480b513366fa3 staging: unisys: visorbus: remove unused long long bus_count 
commit 780f344cdcc89962e9ab6dbe1a3122f893bb8fbb taging: unisys: visorbus: remove unused dump_vhba_bus

selltc, 2016-09-21:
Greg committed these to staging-next on 9/20/2016:
* commit fbd5232109491118eba16773ea2db39b223edde3 - staging: unisys: visorbus: remove unused long long bus_count 
* commit 820b11b6bb4924598524750201cf9360013fe7b7 -  staging: unisys: visorbus: remove unused dump_vhba_bus

That should be all the patches needed for this issue.

closed 2016-09-21 by selltc