davidker/unisys #84: redundant definitions gregkh

arfvidsonUIS, 2016-08-26:
greg kh
>  
static DEFINE_SPINLOCK(parahotplug_request_list_lock); / lock for above / static void parahotplug_process_list(void);  definition not needed, you define it before you use it.

arfvidsonUIS, 2016-08-26:
I have pushed my fix to this issue to branch githubissue-84. I will continue fixing other fixes and hopefully I notice more of this issues elsewhere

davidker, 2016-09-19:
@selltc, does db2bf95e36469b8dcaf6bb6678fb6be6258e68a3 address this issue from greg?

selltc, 2016-09-19:
The original description somehow got mangled, so I just now fixed it.

Looking at it now, my read is that Greg is simply complaining that the prototype for `parahotplug_process_list(void)` was extraneous.  Given that, I think the issue was fixed when the prototype was removed, which I suspect was an earlier patch.

selltc, 2016-09-21:
David pointed out that the "earlier patch" has already been accepted by Greg into staging-next.  It is commit a9c73937436a412aed24c08595a31b7bf9e10dd6 (staging: unisys: visorbus: move parahotplug_process_list), and was accepted by Greg into staging-next 9/20.  This commit was part of githhub issue #88 (Reorder function prototypes), which is now closed.

Given that, I am closing this too.

closed 2016-09-21 by selltc