davidker/unisys #85: Cleanup include files.

davidker, 2016-08-29:
Looks at visorbus.h files after the move is visorbus.h still use " or should it use <>?

 > #include "visorbus.h"
 > #include "visorbus_private.h"
 > #include "vmcallinterface.h"
 
 That's loads of "private" .h files, are they all really needed?

Github issues #68, #63, #67, #76 and revisit.

Added to KanBoard

davidker, 2016-08-29:
created branch [githubissue-86-upstream](https://github.com/davidker/unisys/commits/githubissue-86-upstream) with the fixes for this.

selltc, 2016-08-29:
I looked at all of your patches in [githubissue-86-upstream](/davidker/unisys/commits/githubissue-86-upstream), and have included all my comments below.

Commit 2937f9a5682318115773807e431ab731ab253255 - staging: unisys: visorbus: Reorder functions to avoid including header
* You should have just removed this line instead of just commenting it out:

  ```
+//#include "visorbus_private.h"
  ```

Commit e20a59dfa811813e90bcdf0dd80cb1b0c8e74713 - staging: unisys: visorbus: merge vbusdeviceinfo.h into vbuschannel 
* Your removal of `#include "vbuschannel.h"` from visorbus_main.c does NOT appear to be directly related to getting rid of `vbusdeviceinfo.h`.  Should this be a separate patch?

Commit fb5640532d867db8f4ede8ef04e957cccf54e724 - staging: unisys: visorbus: Merge iovmcall into vmcallinterface 
* Your description should say `iovmcall_gnuc`, instead of just `iovmcall`.

Commit a43896f75c97d2d2bfb03058080b80fefd43bd64 - staging: unisys: visorbus: merge guestlinuxdebug with vmcallinterface.h
* This patch also gets rid of `#ifdef __GNUC__` -- should that be a separate patch?

```

```

davidker, 2016-08-30:
Tim, I updated the branch and renamed it to the proper github issues with your comments. Do you really want a Signed-off-by or do you want Reviewed-by: ? 

I also need to figure out how to do that automatically.

selltc, 2016-08-30:
Oops... you're right, I meant:
```
Reviewed-by: Tim Sell <Timothy.Sell@unisys.com>
```

davidker, 2017-01-17:
At least one of the files has been included and it appears Greg accepted them all. I think there are still more issues but we will uncover them as we so more cleanup. One of the commits that is f6573adedb7d8ee4be3a88fdb2b2e944695946d7

closed 2017-01-17 by davidker