davidker/unisys #88: Reorder function prototypes

davidker, 2016-09-01:
Shouldn't need all these function prototypes reorder functions to get rid of them. 
 > 
 > /* prototypes for attributes */
 > static ssize_t toolaction_show(struct device *dev,
 > 			       struct device_attribute *attr, char *buf);
 > static ssize_t toolaction_store(struct device *dev,
 > 				struct device_attribute *attr,
 > 				const char *buf, size_t count);
 
 Both prototypes not needed.
 
 > static DEVICE_ATTR_RW(toolaction);
 
 Put this below the functions below.
 
 > 
 > static ssize_t boottotool_show(struct device *dev,
 > 			       struct device_attribute *attr, char *buf);
 > static ssize_t boottotool_store(struct device *dev,
 > 				struct device_attribute *attr, const char *buf,
 > 				size_t count);
 > static DEVICE_ATTR_RW(boottotool);
 > 
 > static ssize_t error_show(struct device *dev, struct device_attribute *attr,
 > 			  char *buf);
 > static ssize_t error_store(struct device *dev, struct device_attribute *attr,
 > 			   const char *buf, size_t count);
 > static DEVICE_ATTR_RW(error);
 > 
 > static ssize_t textid_show(struct device *dev, struct device_attribute *attr,
 > 			   char *buf);
 > static ssize_t textid_store(struct device *dev, struct device_attribute *attr,
 > 			    const char *buf, size_t count);
 > static DEVICE_ATTR_RW(textid);
 > 
 > static ssize_t remaining_steps_show(struct device *dev,
 > 				    struct device_attribute *attr, char *buf);
 > static ssize_t remaining_steps_store(struct device *dev,
 > 				     struct device_attribute *attr,
 > 				     const char *buf, size_t count);
 > static DEVICE_ATTR_RW(remaining_steps);
 > 
 > static ssize_t devicedisabled_store(struct device *dev,
 > 				    struct device_attribute *attr,
 > 				    const char *buf, size_t count);
 > static DEVICE_ATTR_WO(devicedisabled);
 > 
 > static ssize_t deviceenabled_store(struct device *dev,
 > 				   struct device_attribute *attr,
 > 				   const char *buf, size_t count);
 > static DEVICE_ATTR_WO(deviceenabled);
 
 Odds are all of these prototypes aren't needed, then:
 
 > 
 > static struct attribute *visorchipset_install_attrs[] = {
 > 	&dev_attr_toolaction.attr,
 > 	&dev_attr_boottotool.attr,
 > 	&dev_attr_error.attr,
 > 	&dev_attr_textid.attr,
 > 	&dev_attr_remaining_steps.attr,
 > 	NULL
 > };
 
 You can move that lower.

... 
 > 
 > /* Function prototypes */
 > static void controlvm_respond(struct controlvm_message_header *msg_hdr,
 > 			      int response);
 > static void controlvm_respond_chipset_init(
 > 		struct controlvm_message_header *msg_hdr, int response,
 > 		enum ultra_chipset_feature features);
 > static void controlvm_respond_physdev_changestate(
 > 		struct controlvm_message_header *msg_hdr, int response,
 > 		struct spar_segment_state state);
 > 
 > static void parser_done(struct parser_context *ctx);
 
 I doubt you need all of those prototypes.

davidker, 2016-09-09:
Updated githubissue-88-upstream-next branch with the changes. Note: There are some patches at the end of the series that isn't related to this issue, I need to clean that up. I just got busy trying to clean things up.

selltc, 2016-09-11:
As you requested via IM, I will take a look at the most-recent 18 commits in [githubissue-88-upstream-next](/davidker/unisys/commits/githubissue-88-upstream-next), which are:
* commit 13d2b24 staing: unisys: include: RRemove unused ULTRA_CHANNELCLI_STRING
* commit f95be7a staging: unisys: include: Remove unused spar_channel_client_release_os
* commit 6f2fb2d staging: unisys: visorbus: remove unused spar_channel_client_acquire_os
* commit ee928a6 staging: unisys: include Remove unused function pathname_last_n_nodes
* commit 5ce633d staging: unisys: visorhba: remove prototypes for visorhba_main.c
* commit 799883b staging: unisys: visornic: remove poll_for_irq
* commit 3274e7f staging: unisys: visornic: remove visornic_poll_for_irq
* commit c94bddf staging: unisys: visornic: remove visornic debugfs protoyptes
* commit 4ff3488 staging: unisys: visornic: remove prototypes
* commit 744d258 staging: unisys: visorbus: move DEFINE SPINLOCK
* commit 156facf staging: unisys: visorbus: remove fix_vbus_dev_info prototypes
* commit 5e9acaf staging: unisys: move visorbus_uevent to remove prototypes
* commit 6af52f5 staging: unisys: visorbus: move devicedisabled/enabled
* commit 4ff5d66 staging: unisys: visorbus: move deviceenabled/disabled functions
* commit 178309e staging: unisys: visorbus: Moved the show/store remaining step functions
* commit e9ed4562 staging: unisys: visorbus: move textid store and show functions
* commit c50b676 staging: unisys: visorbus: move error store/show functions
* commit de197b7 staging: unisys: visorbus: move boottotool

It looks to me like there are a handful of earlier patches that are also related to this issue, but I will ignore those for now.  I'll make my comments directly in the commits themselves, and indicate below which commits I commented on.

selltc, 2016-09-11:
This is my review of the oldest-9 commits; I will provide another comment to review the remaining commits.  All of your code looks good, I just have a few minor nits with a few of the comments.
* commit de197b7 staging: unisys: visorbus: move boottotool - ok

* commit c50b676 staging: unisys: visorbus: move error store/show functions:

    *frunctions* --> *functions*

* commit e9ed45624089b8a8095a8f3c58716e21fce15b09 staging: unisys: visorbus: move textid store and show functions - ok

* commit 178309e staging: unisys: visorbus: Moved the show/store remaining step functions

    *Moved the show/store remaining step functions* -->
    *Moved the remaining_steps_show/store functions*

* commit 4ff5d66 staging: unisys: visorbus: move deviceenabled/disabled functions

    *Move the device/enabled show and store functions need to be remove function
prototypes* -->
    *Move the deviceenabled/devicedisabled show and store functions so that function prototypes can be removed*

* commit 6af52f5 staging: unisys: visorbus: move devicedisabled/enabled

*for the devicedisabled/enable sysfs entries* -->
*for the devicedisabled/deviceenabled sysfs entries*

* commit 5e9acaf staging: unisys: move visorbus_uevent to remove prototypes

    *move visorbus_uevent to remove prototypes* -->
    *move visorbus_type to remove prototypes*

* commit 156facf staging: unisys: visorbus: remove fix_vbus_dev_info prototypes

    *remove fix_vbus_dev_info prototypes* -->
    *remove fix_vbus_dev_info prototype* -->

* commit 744d258 staging: unisys: visorbus: move DEFINE SPINLOCK

    ```
staging: unisys: visorbus: move DEFINE SPINLOCK
DEFINE_SPINLOCK should be defined closer to where we are actually using it.
    ```
    -->
    ```
staging: unisys: visorbus: move parahotplug_request_list and related lock
parahotplug_request_list and parahotplug_request_list_lock should be
defined closer to where we are actually using them.
    ```

  
```
Reviewed-by: Tim Sell <Timothy.Sell@unisys.com>
```

selltc, 2016-09-11:
This is my review of the remaining (newest) 9 commits, so this completes my review of the patches you requested.  As before, all of your code looks good, I just have a few minor nits with a few of the comments.

* commit 4ff3488 staging: unisys: visornic: remove prototypes - ok

* commit c94bddf staging: unisys: visornic: remove visornic debugfs protoyptes - ok

* commit 3274e7f staging: unisys: visornic: remove visornic_poll_for_irq

    ```
staging: unisys: visornic: remove visornic_poll_for_irq
Remove visornic_poll_for_irq prototype, not needed.
    ```
    -->
    ```
staging: unisys: visornic: remove visornic_poll prototype
Remove visornic_poll prototype, because it is not needed.
    ```
* commit 799883b staging: unisys: visornic: remove poll_for_irq

    *staging: unisys: visornic: remove poll_for_irq*
    -->
    *staging: unisys: visornic: remove poll_for_irq prototype*

* commit 5ce633d staging: unisys: visorhba: remove prototypes for visorhba_main.c - ok

* commit ee928a6 staging: unisys: include Remove unused function pathname_last_n_nodes

    *staging: unisys: include Remove unused function pathname_last_n_nodes*
    -->
    *staging: unisys: include: remove unused function pathname_last_n_nodes*

* commit 6f2fb2d staging: unisys: visorbus: remove unused spar_channel_client_acquire_os - ok

* commit f95be7a staging: unisys: include: Remove unused spar_channel_client_release_os

    ```
staging: unisys: include: Remove unused spar_channel_client_release_os
The function spar_channel_client_release_os string is not being
used so remove it.
    ```
    -->
    ```
staging: unisys: include: remove unused spar_channel_client_release_os
The function spar_channel_client_release_os is not being
used so remove it.
    ```

* commit 13d2b24 staing: unisys: include: RRemove unused ULTRA_CHANNELCLI_STRING

    ```
staing: unisys: include: RRemove unused ULTRA_CHANNELCLI_STRING
The define ULTRA_CHANNELCLI_STRING is not being used so remove it.
    ```
    -->
    ```
staging: unisys: include: remove unused ULTRA_CHANNELCLI_STRING
The inline ULTRA_CHANNELCLI_STRING is not being used so remove it.
    ```

```
Reviewed-by: Tim Sell <Timothy.Sell@unisys.com>
```

davidker, 2016-09-12:
Tim can you look at the patches: 221803f782fb0988c0bca658a84f963608a595fc..e8e277768813f82d6a80f65a3ba3c5fc243cda2e, I miscounted when I asked you. 

Thanks!

selltc, 2016-09-12:
Sure; I'll look at these next:
* commit e8e2777 staging: unisys: visorbus: move toolaction functions to remove prototypes
* commit 6839dd7 staging: unisys: visorbus: remove parser_done
* commit ae6b674 staging: unisys: visorbus: remove controlvm_respond_physdev_changestate
* commit 4f69553 staging: unisys: visorbus: remove controlvm_respond_chipset_init prototype
* commit 6092056 staging: unisys: visorbus: remove prototype controlvm_respond
* commit 9853349 staging: unisys: visorbus: move parahotplug_process_list
* commit 4f43fea staging: unisys: visorbus: move read_controlvm_event
* commit 4cdfb46 staging: unisys: visorbus: move parser_init_stream
* commit 8f64fed staging: unisys: visorbus: Move handle_command
* commit 221803f staging: unisys: visorbus: Reorder controlvm_periodic_work

selltc, 2016-09-12:
This is my review of the remaining (oldest) 10 commits pointed out in the previous comment, so this completes my review of the patches you requested. As before, all of your code looks good, I just have a few minor nits with a few of the comments.

* commit 221803f staging: unisys: visorbus: Reorder controlvm_periodic_work

    *Move controlvmc_periodic_work lower*
    -->
    *move controlvm_periodic_work lower*

* commit 8f64fed staging: unisys: visorbus: Move handle_command - might want to change "Move" to "move", just to be consistent with all of your other comments

* commit 4cdfb46 staging: unisys: visorbus: move parser_init_stream - ok

* commit 4f43fea staging: unisys: visorbus: move read_controlvm_event - ok

* commit 9853349 staging: unisys: visorbus: move parahotplug_process_list - ok

* commit 6092056 staging: unisys: visorbus: remove prototype controlvm_respond - "uneeded" --> "unneeded"

* commit 4f69553 staging: unisys: visorbus: remove controlvm_respond_chipset_init prototype - "uneeded" --> "unneeded"

* commit ae6b674 staging: unisys: visorbus: remove controlvm_respond_physdev_changestate - "uneeded" --> "unneeded"

* commit 6839dd7 staging: unisys: visorbus: remove parser_done - "remove parser_done" --> "remove parser_done prototype"

* commit e8e2777 staging: unisys: visorbus: move toolaction functions to remove prototypes - ok

selltc, 2016-09-21:
Greg committed these to staging-next on 9/20/2016:
* commit 51319662d6ac5729a1950579d70808d94e4fa8e6 staging: unisys: visorbus: move parahotplug_request_list and related lock
* commit 7a0ee694889694c49d4ebe99dce30e699f832396 staging: unisys: visorbus: remove fix_vbus_dev_info prototype
* commit 7993b40cd772cae9890f81fb0ea809130a732c5e staging: unisys: visorbus move visorbus_type to remove prototypes
* commit ebeff0558c0a311f4c5d432c69c32b9502219190 staging: unisys: visorbus: move deviceenabled/disabled store functions
* commit 97f792ee3cd6f62551f6247ca36492615e26cb5e staging: unisys: visorbus: move the remaining_steps_show/store functions
* commit 79730c7c8148f68144540181b8b53b1e9639f9b7 staging: unisys: visorbus: move textid store and show functions
* commit 8a4a8a03d3a378694ba4170370a447d3e0df3ad9 staging: unisys: visorbus: move error store/show functions
* commit 65cb1a8c5471ee2aed372dbf69a51457d15de056 staging: unisys: visorbus: remove parser_done prototype
* commit dd845d3632135c5e17aaadccddda56f849b03201 staging: unisys: visorbus: remove controlvm_respond_physdev_changestate
* commit 612b81c9315c60f3ab921cf87e1740e087a103de staging: unisys: visorbus: move parser_init_byte_stream
* commit 1b1d463d08be0785b48ad8b23446ac13f726a60b staging: unisys: visorbus: move boottotool
* commit 84efd2077c1e935c4123e5688ba86062f608bb73 staging: unisys: visorbus: move toolaction functions to remove prototypes
* commit 5f2513950f1f8b9acc740720b89389b2aa24ac72 staging: unisys: visorbus: remove controlvm_respond_chipset_init prototype
* commit 992228b2b0f2d7a1efc9cc21c21f1a3dea9c50f5 staging: unisys: visorbus: remove prototype controlvm_respond
* commit a9c73937436a412aed24c08595a31b7bf9e10dd6 staging: unisys: visorbus: move parahotplug_process_list
* commit 8a2853279b3316558d4a5a34b06b6890ebc65613 staging: unisys: visorbus: move read_controlvm_event
* commit 511474a5077b088f92166a7875af5e6fa428e465 staging: unisys: visorbus: move handle_command
* commit 3d8394c86f1b482ec2509e5a152484a27cc6cc74 staging: unisys: visorbus: reorder controlvm_periodic_work

selltc, 2016-09-21:
Greg committed these to staging-next on 9/21/2016:
* commit e7b6d8a61e3244392f882a31d08eab338cdeb772 staging: unisys: visornic: remove poll_for_irq prototype
* commit 2e805a9738a0487db1f5b8b6cb41b6c4cd882b63 staging: unisys: visornic: remove visornic_poll prototype
* commit 0543205b7e216288c7905cedf5ba804f3ee37599 staging: unisys: visornic: remove visornic debugfs protoyptes
* commit 6083c7108193f32da8792fbbce18d80ee4ca75af staging: unisys: visornic: remove prototypes
* commit d616700e38d91f9ec11d83ccf64d8e038935f26d -  staging: unisys: visorhba: remove prototypes for visorhba_main.c

This completes the patches required for this issue.

closed 2016-09-21 by selltc

selltc, 2016-09-21:
Oops... missed a few other patches also committed by Greg to staging-next on 9/21/2016:
* commit a43bb30b62cf6993f1351c3f0d25f048e96052cd staging: unisys: include: remove unused ULTRA_CHANNELCLI_STRING
* commit 0082b016b59e67cbbb747166565074f61255da7d staging: unisys: include: remove unused spar_channel_client_release_os
* commit 207d14a1aa70fd5bafdc3900f431df40c733c8ab staging: unisys: include: remove unused spar_channel_client_acquire_os
* commit 3838925ce22063cb8e2b70311d3163b869bc6592 staging: unisys: include: remove unused function pathname_last_n_nodes