davidker/unisys #97: Remove useless initialization

wadgaonkarsam, 2016-09-19:
[Kanboard 24493](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=24493&project_id=37)

davidker, 2016-09-22:
Great job with the patch! Looks good, though looking at the issue, I'm not sure if your patch will allow us to close the patch. I think there are several other spots in the driver that greg would complain about if he had another eight hour flight to look through the code. We might want to comb through the code and see if we can find other spots or at least comb through his response some more to see if we can find some.

wadgaonkarsam, 2016-09-22:
Thanks David! I'll keep looking for similar initializations in the code. I added back the memset in initialize_controlvm_payload_info().

davidker, 2016-09-28:
Sameer, I picked up your latest patch and started to review it. The patch itself is fine, however, looking at your signed off line your name is all lowercase, are you sure you want that? The signed-off line is like a legal document and your name should be represented as a signature. Do you want me to fix it, or do you want respin?

davidker, 2016-09-28:
Actually, I take it back about the patch itself is fine. You probably should have two patches one to do the removal and one to do the move down in the file.

davidker, 2016-09-28:
The more I think about it, we can probably keep the patch as is, so ignore my previous comment.

wadgaonkarsam, 2016-09-28:
Thank you David, for the inspection and updating the signed off line.

davidker, 2017-01-17:
Committed at  c4fe36cd262d5fe710ff9ee43cb99337e8b050de

closed 2017-01-17 by davidker