davidker/unisys #98: Remove the "local" variable

arfvidsonUIS, 2016-09-19:


arfvidsonUIS, 2016-09-19:
Tasks:
in visorchannel rename it to dest and same with visorbus_private

get rid of local and local_addr in visorchipset 
might be able to get rid of test_message
though if someone does send us a test_message we probably want to ignore it now

arfvidsonUIS, 2016-09-28:
updated branch githubissue-98 and ready for review

arfvidsonUIS, 2016-09-28:
checked by running grep -r -n local drivers/staging/unisys/

davidker, 2016-09-30:
The patch 057c711f8845d14bfdbfb2bd68e1fe737387e6b7fails to compile cleanly when -werror is used on the driver. 

I've accepted the other two patches, but am pulling this one from upstream-next. 

>   LD      init/built-in.o
  CC      kernel/sys.o
  CC      kernel/module.o
  CC      drivers/base/firmware_class.o
  CC      drivers/staging/unisys/visorbus/visorchipset.o
../drivers/staging/unisys/visorbus/visorchipset.c: In function 'parser_init_byte_stream':
../drivers/staging/unisys/visorbus/visorchipset.c:1850:2: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-afte
r-statement]
  void *mapping = memremap(addr, bytes, MEMREMAP_WB);
  ^
  LD      kernel/built-in.o
  LD      drivers/base/built-in.o
cc1: all warnings being treated as errors
make[5]: *** [drivers/staging/unisys/visorbus/visorchipset.o] Error 1
make[4]: *** [drivers/staging/unisys/visorbus] Error 2
make[3]: *** [drivers/staging/unisys] Error 2
make[2]: *** [drivers/staging] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [drivers] Error 2
make[1]: Leaving directory `/home/kershnda/kernel/master/unisys/.build'
make: *** [sub-make] Error 2
FAILED make
Execution failed: ~/bin/make_commit.sh -w
and made changes to the index and/or the working tree
You can fix the problem, and then run

arfvidsonUIS, 2016-10-03:
Cheers,
Erik Arfvidson

From: davidker
Sent: Friday, September 30, 9:32 AM
Subject: Re: [davidker/unisys] Remove the "local" variable (#98)
To: davidker/unisys
Cc: Arfvidson, Erik, Author

The patch 057c711f8845d14bfdbfb2bd68e1fe737387e6b7fails to compile cleanly when -werror is used on the driver.

I've accepted the other two patches, but am pulling this one from upstream-next.

LD init/built-in.o
CC kernel/sys.o
CC kernel/module.o
CC drivers/base/firmware_class.o
CC drivers/staging/unisys/visorbus/visorchipset.o
../drivers/staging/unisys/visorbus/visorchipset.c: In function 'parser_init_byte_stream':
../drivers/staging/unisys/visorbus/visorchipset.c:1850:2: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-afte
r-statement]
void mapping = memremap(addr, bytes, MEMREMAP_WB);
^
LD kernel/built-in.o
LD drivers/base/built-in.o
cc1: all warnings being treated as errors
make[5]: ** [drivers/staging/unisys/visorbus/visorchipset.o] Error 1
make[4]: *** [drivers/staging/unisys/visorbus] Error 2
make[3]: *** [drivers/staging/unisys] Error 2
make[2]: *** [drivers/staging] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [drivers] Error 2
make[1]: Leaving directory `/home/kershnda/kernel/master/unisys/.build'
make: *** [sub-make] Error 2
FAILED make
Execution failed: ~/bin/make_commit.sh -w
and made changes to the index and/or the working tree
You can fix the problem, and then run

????
not sure what you mean here. What would you like me to run?
-

davidker, 2016-10-03:
I need you to make sure there are no warnings when you compile the patch. My script runs with -werror set for our drivers and it gave the instructions at the end of its error. 

You should be getting warnings when you build our drivers with this patch: 

> ../drivers/staging/unisys/visorbus/visorchipset.c: In function 'parser_init_byte_stream':
../drivers/staging/unisys/visorbus/visorchipset.c:1850:2: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-afte
r-statement]
void mapping = memremap(addr, bytes, MEMREMAP_WB);

That needs to be fixed before I can send it upstream.

davidker, 2016-10-03:
Please also change: 

> local_addr = (inmsg.hdr.flags.test_message == 1);

to something like: 

> if ((insmsg.hdr.flags.test_message == 1) &&
     (inmsg.hdr.flags.response_expected)) {
            controlvm_respond
                                (&inmsg.hdr,
                                 -CONTROLVM_RESP_ERROR_MESSAGE_INVALID);
             return true;
}

arfvidsonUIS, 2016-10-03:
change function from 4 to 3 parameters
remove local_addr

arfvidsonUIS, 2016-10-03:
Just updated branch githubissue-98. Ready for submission

closed 2017-01-12 by davidker