davidker/unisys #99: Add default case for all switch statements

TurningWrenches, 2016-09-25:


TurningWrenches, 2016-09-25:
```
> grep -rn "switch (.*)" *
visorbus/visorchipset.c:385:    switch (which_string) {
visorbus/visorchipset.c:730:            switch (cmd) {
visorbus/visorchipset.c:781:            switch (cmd) {
visorbus/visorchipset.c:1700:   switch (offset) {
visorbus/visorchipset.c:1753:   switch (cmd) {
visorbus/visorchipset.c:1932:   switch (inmsg.hdr.id) {
visorhba/visorhba_main.c:739:           switch (pendingdel->cmdtype) {
visorinput/visorinput.c:395:    switch (devtype) {
visorinput/visorinput.c:508:    switch (keycode) {
visorinput/visorinput.c:551:    switch (x) {
visorinput/visorinput.c:587:            switch (r.activity.action) {
visornic/visornic_main.c:1044:  switch (copy.numrcvbufs) {
visornic/visornic_main.c:1573:          switch (cmdrsp->net.type) {
```

TurningWrenches, 2016-09-29:
**Branch:** githubissue-99-staging-next-TurningWrenches
**Commit(s):** 
cc99fdd9a296d6c626c44297705f6e060166d039
**checkpatch-cleared:** Yes.
**T710-1 verified:** No.

davidker, 2016-10-01:
David, I'm going to hold off on submitting this one until I submit the patches to remove the epilog functions. My reasoning is that he complained about the ones in epilog, if we fix visorinput and he remembers he might come back and say why aren't you fixing the ones I complained about. When in reality I am getting rid of it.

TurningWrenches, 2016-10-01:
Sounds great. Thanks!

davidker, 2017-01-17:
Committed to staging-next at a4e93a694877b68b88eb6e06442ad904809eef51

closed 2017-01-17 by davidker