davidker/unisys #100: /sys/bus/visorbus/devices/visorbus<n>/client_bus_info should be a debufs entry

selltc, 2016-09-28:
In visorbus_main.c, /sys/bus/visorbus/devices/visorbus<n>/client_bus_info should be a debugfs entry instead of a sysfs entry, as per Greg's comment when we recently-submitted a change to `client_bus_info_show()` (see github #90):

    You do realize that sysfs files are supposed to only be "one value per
    file", right? This is going to have to change soon, if you want this
    code out of staging.

    I'll take these patches now, but please work on a sane API (this really
    looks like something for debugfs as no userspace tools should be relying
    on it...)

selltc, 2016-09-28:
Fixes on [githubissue-100-upstream-next branch](/davidker/unisys/commits/githubissue-100-upstream-next).

These are checkpatch-clean and tested successfully.

selltc, 2016-09-30:
Proposed cover letter:
```
Move /sys/bus/visorbus/devices/visorbus<n>/client_bus_info to debugfs

In a 9/28/2016 patch inspection that fixed code within the function that
emits this client_bus_info sysfs entry, GregKH pointed out that this
shouldn't actually be a sysfs entry at all, and in fact belongs in debugfs.
This patchset addresses that.
```

selltc, 2016-09-30:
I also successfully compiled this code using a .config where `CONFIG_DEBUG_FS` is NOT set, just to be sure I didn't break anything there.

davidker, 2017-02-28:
8217becc1b416e348d1bc8fcb54c09967e0f4987 fixes this

closed 2017-02-28 by davidker