davidker/unisys #102: Shorten error message defines

wadgaonkarsam, 2016-09-28:
[Kanboard:#24536](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=24536&project_id=37)

 ```
 		pmsg_hdr = msg_hdr;
 		goto out_respond;
 	}
 
 	if (bus_info->pending_msg_hdr) {
 		/* only non-NULL if dev is still waiting on a response */
 		response = -CONTROLVM_RESP_ERROR_MESSAGE_ID_INVALID_FOR_CLIENT;
```
 
 I_LIKE_LONG_ERROR_MESSAGE_DEFINES_AND_I_CAN_NOT_LIE
 
 Ok, I lie...
 
 ```
	bus_info = visorbus_get_device_by_id(bus_no, BUS_ROOT_DEVICE, NULL);
 	if (!bus_info) {
 		POSTCODE_LINUX_3(BUS_CONFIGURE_FAILURE_PC, bus_no,
 				 POSTCODE_SEVERITY_ERR);
 		rc = -CONTROLVM_RESP_ERROR_BUS_INVALID;
 	} else if (bus_info->state.created == 0) {
 		POSTCODE_LINUX_3(BUS_CONFIGURE_FAILURE_PC, bus_no,
 				 POSTCODE_SEVERITY_ERR);
 		rc = -CONTROLVM_RESP_ERROR_BUS_INVALID;
 	} else if (bus_info->pending_msg_hdr) {
 		POSTCODE_LINUX_3(BUS_CONFIGURE_FAILURE_PC, bus_no,
 				 POSTCODE_SEVERITY_ERR);
 		rc = -CONTROLVM_RESP_ERROR_MESSAGE_ID_INVALID_FOR_CLIENT;
```
 This is just a bad dream, right?

wadgaonkarsam, 2016-11-21:
**Branch:** githubissue-109-for-later-TurningWrenches
**Commit(s):** 
d4358eb0d9670a7d231b2c51afffdec9e13ba250
**checkpatch-cleared:** Yes
**T710-W1 verified:** Yes

wadgaonkarsam, 2016-11-28:
Updated the branch with upstream-next
Commits: 68a0cea17a489b6076ecf17919c13e33b1cd4360 59dffa2c90c91e93dd58f57caf953a54013046f3

davidker, 2017-01-19:
The two patches got squashed and committed to staging-next at: 98f9ed9e

closed 2017-01-19 by davidker