davidker/unisys #103: Remove useless check

TurningWrenches, 2016-09-30:
``` 
 > 	} else {
 > 		void *mapping = memremap(addr, bytes, MEMREMAP_WB);
 > 
 > 		if (!mapping)
 > 			goto err_finish_ctx;
 > 		memcpy(ctx->data, mapping, bytes);
 > 		memunmap(mapping);
 > 	}
 > 
 > 	ctx->byte_stream = true;
 > 	controlvm_payload_bytes_buffered += ctx->param_bytes;
 > 
 > 	return ctx;
 > 
 > err_finish_ctx:
 > 	parser_done(ctx);
 > 	return NULL;
 > }
 > 
 > static uuid_le
 > parser_id_get(struct parser_context *ctx)
 > {
 > 	struct spar_controlvm_parameters_header *phdr = NULL;
 > 
 > 	if (!ctx)
 > 		return NULL_UUID_LE;
 ```

 How can that ever be true?
...

```
 > 	return phdr->id;
 > }
 > 
 > /*
 >  * Describes the state from the perspective of which controlvm messages have
 >  * been received for a bus or device.
 >  */
 > 
 > enum PARSER_WHICH_STRING {
 > 	PARSERSTRING_INITIATOR,
 > 	PARSERSTRING_TARGET,
 > 	PARSERSTRING_CONNECTION,
 > 	PARSERSTRING_NAME, /* TODO: only PARSERSTRING_NAME is used ? */
 > };
 > 
 > static void
 > parser_param_start(struct parser_context *ctx,
 > 		   enum PARSER_WHICH_STRING which_string)
 > {
 > 	struct spar_controlvm_parameters_header *phdr = NULL;
 > 
 > 	if (!ctx)
 > 		return;
 ```

 How could that happen?

...

```
 > 	default:
 > 		break;
 > 	}
 > }
 > 
 > static void parser_done(struct parser_context *ctx)
 > {
 > 	if (!ctx)
 > 		return;
 ```

 How can this happen?

... 

```
 > 	controlvm_payload_bytes_buffered -= ctx->param_bytes;
 > 	kfree(ctx);
 > }
 > 
 > static void *
 > parser_string_get(struct parser_context *ctx)
 > {
 > 	u8 *pscan;
 > 	unsigned long nscan;
 > 	int value_length = -1;
 > 	void *value = NULL;
 > 	int i;
 > 
 > 	if (!ctx)
 > 		return NULL;
 ```

 Again, how?

TurningWrenches, 2016-09-30:
**Branch:**
**Commit(s):** 
**checkpatch-cleared:**
**T710-1 verified:**

TurningWrenches, 2016-10-02:
```
$ grep -rn 'if (!ctx)' *
visorbus/visorchipset.c:356:    if (!ctx)
visorbus/visorchipset.c:380:    if (!ctx)
visorbus/visorchipset.c:408:    if (!ctx)
visorbus/visorchipset.c:423:    if (!ctx)
visorbus/visorchipset.c:1839:   if (!ctx) {
```

TurningWrenches, 2016-10-02:
Reminder: do a double-check that the changed code are in fact non-API functions.

TurningWrenches, 2016-10-03:
**Branch:** githubissue-103-staging-next-TurningWrenches
**Commit(s):** 
e2b8be31fe978e88a4e2dfcb195ab8d3cfc8c930
65b16ed2de1e4ee42b5ef93a0544407d4c84749f
65e469e1ed9926a7406895b50e24168f8f11bd27
**checkpatch-cleared:** No.
**T710-1 verified:** No.

TurningWrenches, 2016-11-19:
-+-+-+-+-+-+
~ ROUND 2 ~
-+-+-+-+-+-+

```
$ grep -rn "if (\!.*)" drivers/staging/unisys 
drivers/staging/unisys/visorbus/visorchannel.c:61:	if (!channel)
drivers/staging/unisys/visorbus/visorchannel.c:322:	if (!channel->needs_lock)
drivers/staging/unisys/visorbus/visorchannel.c:412:	if (!channel)
drivers/staging/unisys/visorbus/visorchannel.c:427:	if (!channel->requested) {
drivers/staging/unisys/visorbus/visorchannel.c:435:	if (!channel->mapped) {
drivers/staging/unisys/visorbus/visorchannel.c:460:	if (!channel->requested) {
drivers/staging/unisys/visorbus/visorchannel.c:469:	if (!channel->mapped) {

drivers/staging/unisys/visorbus/visorbus_main.c:120:	if (!drv->channel_types)
drivers/staging/unisys/visorbus/visorbus_main.c:188:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:200:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:212:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:225:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:238:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:254:	if (!vdev->visorchannel || !xbus || !xdrv)
drivers/staging/unisys/visorbus/visorbus_main.c:257:	if (!i)
drivers/staging/unisys/visorbus/visorbus_main.c:391:	if (!channel)
drivers/staging/unisys/visorbus/visorbus_main.c:461:	if (!dev->timer_active)
drivers/staging/unisys/visorbus/visorbus_main.c:642:	 *              if (!drv.probe(dev))  [visordriver_probe_device]
drivers/staging/unisys/visorbus/visorbus_main.c:676:	if (!SPAR_VBUS_CHANNEL_OK_CLIENT(visorchannel_get_header(chan)))
drivers/staging/unisys/visorbus/visorbus_main.c:793:	if (!visordev->device.driver)
drivers/staging/unisys/visorbus/visorbus_main.c:797:	if (!bdev)
drivers/staging/unisys/visorbus/visorbus_main.c:800:	if (!hdr_info)
drivers/staging/unisys/visorbus/visorbus_main.c:856:	if (!drv->probe)
drivers/staging/unisys/visorbus/visorbus_main.c:941:	 *           if (!dev.drv)  ** no driver assigned yet **
drivers/staging/unisys/visorbus/visorbus_main.c:944:	 *               if (!drv.probe(dev))   [visordriver_probe_device]
drivers/staging/unisys/visorbus/visorbus_main.c:972:	if (!hdr_info)
drivers/staging/unisys/visorbus/visorbus_main.c:982:	if (!dev->debugfs_dir) {
drivers/staging/unisys/visorbus/visorbus_main.c:990:	if (!dev->debugfs_client_bus_info) {
drivers/staging/unisys/visorbus/visorbus_main.c:1157:	if (!dev->pausing)
drivers/staging/unisys/visorbus/visorbus_main.c:1177:	if (!dev->resuming)
drivers/staging/unisys/visorbus/visorbus_main.c:1212:	if (!notify_func)
drivers/staging/unisys/visorbus/visorbus_main.c:1216:	if (!drv) {
drivers/staging/unisys/visorbus/visorbus_main.c:1235:		if (!drv->pause) {
drivers/staging/unisys/visorbus/visorbus_main.c:1250:		if (!drv->resume) {
drivers/staging/unisys/visorbus/visorbus_main.c:1303:	if (!visorbus_debugfs_dir)

drivers/staging/unisys/visorbus/visorchipset.c:356:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:380:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:408:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:423:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:429:	if (!pscan)
drivers/staging/unisys/visorbus/visorchipset.c:439:	if (!value)
drivers/staging/unisys/visorbus/visorchipset.c:646:	if (!pending_msg_hdr)
drivers/staging/unisys/visorbus/visorchipset.c:664:	if (!p->pending_msg_hdr)
drivers/staging/unisys/visorbus/visorchipset.c:684:	if (!pending_msg_hdr)
drivers/staging/unisys/visorbus/visorchipset.c:712:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:731:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:749:	if (!visorchannel) {
drivers/staging/unisys/visorbus/visorchipset.c:785:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:800:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:837:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:890:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:913:	if (!dev_info) {
drivers/staging/unisys/visorbus/visorchipset.c:936:	if (!visorchannel) {
drivers/staging/unisys/visorbus/visorchipset.c:950:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:985:	if (!dev_info) {
drivers/staging/unisys/visorbus/visorchipset.c:1004:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:1045:	if (!dev_info) {
drivers/staging/unisys/visorbus/visorchipset.c:1061:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:1099:	if (!info)
drivers/staging/unisys/visorbus/visorchipset.c:1106:	if (!payload)
drivers/staging/unisys/visorbus/visorchipset.c:1211:	if (!req)
drivers/staging/unisys/visorbus/visorchipset.c:1422:	if (!req)
[... 47 lines cut ...]
drivers/staging/unisys/visornic/visornic_main.c:1780:	if (!devdata->cmdrsp_rcv) {
drivers/staging/unisys/visornic/visornic_main.c:1785:	if (!devdata->xmit_cmdrsp) {
drivers/staging/unisys/visornic/visornic_main.c:1855:	if (!devdata->eth_debugfs_dir) {
drivers/staging/unisys/visornic/visornic_main.c:1917:	if (!devdata) {
drivers/staging/unisys/visornic/visornic_main.c:1930:	if (!netdev) {
drivers/staging/unisys/visornic/visornic_main.c:1991:	if (!devdata) {
drivers/staging/unisys/visornic/visornic_main.c:2005:	if (!devdata->server_down) {
drivers/staging/unisys/visornic/visornic_main.c:2058:	if (!visornic_debugfs_dir)
drivers/staging/unisys/visornic/visornic_main.c:2063:	if (!ret)
drivers/staging/unisys/visornic/visornic_main.c:2067:	if (!ret)

[... cut ...]
drivers/staging/unisys/visorhba/visorhba_main.c:973:	if (!cmdrsp)
drivers/staging/unisys/visorhba/visorhba_main.c:1025:	if (!devdata)
drivers/staging/unisys/visorhba/visorhba_main.c:1057:	if (!scsihost)
drivers/staging/unisys/visorhba/visorhba_main.c:1086:	if (!devdata->debugfs_dir) {
drivers/staging/unisys/visorhba/visorhba_main.c:1094:	if (!devdata->debugfs_info) {
drivers/staging/unisys/visorhba/visorhba_main.c:1151:	if (!devdata)
drivers/staging/unisys/visorhba/visorhba_main.c:1192:	if (!visorhba_debugfs_dir)
```

TurningWrenches, 2016-11-29:
++-+-+-+-+-+
\+  ROUND 2  +
++-+-+-+-+-+

$ grep -rn "if (\\!.*)" drivers/staging/unisys 
drivers/staging/unisys/visorbus/visorchannel.c:61:	if (!channel)
drivers/staging/unisys/visorbus/visorchannel.c:322:	if (!channel->needs_lock)
drivers/staging/unisys/visorbus/visorchannel.c:412:	if (!channel)
drivers/staging/unisys/visorbus/visorchannel.c:427:	if (!channel->requested) {
drivers/staging/unisys/visorbus/visorchannel.c:435:	if (!channel->mapped) {
drivers/staging/unisys/visorbus/visorchannel.c:460:	if (!channel->requested) {
drivers/staging/unisys/visorbus/visorchannel.c:469:	if (!channel->mapped) {

drivers/staging/unisys/visorbus/visorbus_main.c:120:	if (!drv->channel_types)
drivers/staging/unisys/visorbus/visorbus_main.c:188:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:200:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:212:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:225:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:238:	if (!vdev->visorchannel)
drivers/staging/unisys/visorbus/visorbus_main.c:254:	if (!vdev->visorchannel || !xbus || !xdrv)
drivers/staging/unisys/visorbus/visorbus_main.c:257:	if (!i)
drivers/staging/unisys/visorbus/visorbus_main.c:391:	if (!channel)
drivers/staging/unisys/visorbus/visorbus_main.c:461:	if (!dev->timer_active)
drivers/staging/unisys/visorbus/visorbus_main.c:680:	 *              if (!drv.probe(dev))  [visordriver_probe_device]
drivers/staging/unisys/visorbus/visorbus_main.c:714:	if (!SPAR_VBUS_CHANNEL_OK_CLIENT(visorchannel_get_header(chan)))
drivers/staging/unisys/visorbus/visorbus_main.c:831:	if (!visordev->device.driver)
drivers/staging/unisys/visorbus/visorbus_main.c:835:	if (!bdev)
drivers/staging/unisys/visorbus/visorbus_main.c:838:	if (!hdr_info)
drivers/staging/unisys/visorbus/visorbus_main.c:894:	if (!drv->probe)
drivers/staging/unisys/visorbus/visorbus_main.c:979:	 *           if (!dev.drv)  ** no driver assigned yet **
drivers/staging/unisys/visorbus/visorbus_main.c:982:	 *               if (!drv.probe(dev))   [visordriver_probe_device]
drivers/staging/unisys/visorbus/visorbus_main.c:1011:	if (!hdr_info)
drivers/staging/unisys/visorbus/visorbus_main.c:1021:	if (!dev->debugfs_dir) {
drivers/staging/unisys/visorbus/visorbus_main.c:1029:	if (!dev->debugfs_client_bus_info) {
drivers/staging/unisys/visorbus/visorbus_main.c:1200:	if (!dev->pausing)
drivers/staging/unisys/visorbus/visorbus_main.c:1220:	if (!dev->resuming)
drivers/staging/unisys/visorbus/visorbus_main.c:1255:	if (!notify_func)
drivers/staging/unisys/visorbus/visorbus_main.c:1259:	if (!drv) {
drivers/staging/unisys/visorbus/visorbus_main.c:1278:		if (!drv->pause) {
drivers/staging/unisys/visorbus/visorbus_main.c:1293:		if (!drv->resume) {
drivers/staging/unisys/visorbus/visorbus_main.c:1347:	if (!visorbus_debugfs_dir)

drivers/staging/unisys/visorbus/visorchipset.c:356:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:380:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:408:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:423:	if (!ctx)
drivers/staging/unisys/visorbus/visorchipset.c:429:	if (!pscan)
drivers/staging/unisys/visorbus/visorchipset.c:439:	if (!value)
drivers/staging/unisys/visorbus/visorchipset.c:652:	if (!pending_msg_hdr)
drivers/staging/unisys/visorbus/visorchipset.c:670:	if (!p->pending_msg_hdr)
drivers/staging/unisys/visorbus/visorchipset.c:690:	if (!pending_msg_hdr)
drivers/staging/unisys/visorbus/visorchipset.c:718:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:738:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:756:	if (!visorchannel) {
drivers/staging/unisys/visorbus/visorchipset.c:793:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:808:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:845:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:898:	if (!bus_info) {
drivers/staging/unisys/visorbus/visorchipset.c:924:	if (!dev_info) {
drivers/staging/unisys/visorbus/visorchipset.c:948:	if (!visorchannel) {
drivers/staging/unisys/visorbus/visorchipset.c:963:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:998:	if (!dev_info) {
drivers/staging/unisys/visorbus/visorchipset.c:1019:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:1060:	if (!dev_info) {
drivers/staging/unisys/visorbus/visorchipset.c:1076:		if (!pmsg_hdr) {
drivers/staging/unisys/visorbus/visorchipset.c:1114:	if (!info)
drivers/staging/unisys/visorbus/visorchipset.c:1121:	if (!payload)
drivers/staging/unisys/visorbus/visorchipset.c:1228:	if (!req)
drivers/staging/unisys/visorbus/visorchipset.c:1439:	if (!req)
drivers/staging/unisys/visorbus/visorchipset.c:1569:	if (!VMCALL_SUCCESSFUL(issue_vmcall_io_controlvm_addr(&addr, &size)))
[... 45 lines cut ...]
drivers/staging/unisys/visornic/visornic_main.c:1759:	if (!devdata->rcvbuf) {
drivers/staging/unisys/visornic/visornic_main.c:1780:	if (!devdata->cmdrsp_rcv) {
drivers/staging/unisys/visornic/visornic_main.c:1785:	if (!devdata->xmit_cmdrsp) {
drivers/staging/unisys/visornic/visornic_main.c:1855:	if (!devdata->eth_debugfs_dir) {
drivers/staging/unisys/visornic/visornic_main.c:1917:	if (!devdata) {
drivers/staging/unisys/visornic/visornic_main.c:1930:	if (!netdev) {
drivers/staging/unisys/visornic/visornic_main.c:1991:	if (!devdata) {
drivers/staging/unisys/visornic/visornic_main.c:2005:	if (!devdata->server_down) {
[... cut ...]
drivers/staging/unisys/visorhba/visorhba_main.c:973:	if (!cmdrsp)
drivers/staging/unisys/visorhba/visorhba_main.c:1025:	if (!devdata)
drivers/staging/unisys/visorhba/visorhba_main.c:1057:	if (!scsihost)
drivers/staging/unisys/visorhba/visorhba_main.c:1086:	if (!devdata->debugfs_dir) {
drivers/staging/unisys/visorhba/visorhba_main.c:1094:	if (!devdata->debugfs_info) {
drivers/staging/unisys/visorhba/visorhba_main.c:1151:	if (!devdata)
drivers/staging/unisys/visorhba/visorhba_main.c:1192:	if (!visorhba_debugfs_dir)

davidker, 2017-07-27:
David, you have stale branches for these (i.e. > 4 months). Do you want me to keep the branches or do we start fresh? What is your take on that.