davidker/unisys #105: Cleanup iochannel.h

arfvidsonUIS, 2016-10-06:
Fix formatting and comments

arfvidsonUIS, 2016-10-06:
lol hope greg does not get angry with my patch. I'll split it if he complains. Patch checkpatch and builds cleanly. The patch is ready for submission in branch githubissue-105.

davidker, 2016-10-07:
I think this patch really needs to be split up, I'll pre-empt Greg on that one:

The other thing I noticed is that it follows the /net comment block convention instead of the standard Linux convention. Although a large portion of the file could be considered /net code, the file itself currently wants to live in the /include directory, so I think it should follow the standard linux convention. 

Here is the relevant portion of the coding style: 

> Linux style for comments is the C89 "/* ... */" style.
Don't use C99-style "// ..." comments.

The preferred style for long (multi-line) comments is:

	/*
	 * This is the preferred style for multi-line
	 * comments in the Linux kernel source code.
	 * Please use it consistently.
	 *
	 * Description:  A column of asterisks on the left side,
	 * with beginning and ending almost-blank lines.
	 */

For files in net/ and drivers/net/ the preferred style for long (multi-line)
comments is a little different.

	/* The preferred comment style for files in net/ and drivers/net
	 * looks like this.
	 *
	 * It is nearly the same as the generally preferred comment style,
	 * but there is no initial almost-blank line.
	 */

It's also important to comment data, whether they are basic types or derived
types.  To this end, use just one data declaration per line (no commas for
multiple data declarations).  This leaves you room for a small comment on each
item, explaining its use.

arfvidsonUIS, 2016-10-28:
I just split this patch and is ready for submission on branch githubissue-105-v1

davidker, 2017-07-27:
Closing this due to patches not being needed anymore, code churn (part of the patches as well) have fixed this.

closed 2017-07-27 by davidker