davidker/unisys #106: KB25101

TurningWrenches, 2016-10-17:
[KanBoard-25101](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=25101&project_id=37)

TurningWrenches, 2016-10-17:
**Patch Board**

_Relocataion of visobus_request_irq() and visorbus_free_irq()_
bbb1686c82c6cd2b4e8ef9019a8a1ad631b05bf8
b7c7001e7576966d432de6eb05d0b09720f049b5

selltc, 2016-11-05:
I successfully verified David's fix for on my T110 as described [in comment-23 of Trac-7234](https://ustr-svn-1.na.uis.unisys.com/trac/spar/ticket/7234#comment:23).

TurningWrenches, 2016-11-19:
Please also add this commit: fc44a49d292a00284972f79b7c83d378727bf5f1 ( @davidker, @selltc).

selltc, 2016-11-29:
As in #107, we also need to add commit 34801316e9131ad2ac05dde83e5ec1f8c179e2c4 (staging: unisys: remove redundant unlocks of visornic_devdata.priv_lock) here, in order to fix a problem introduced by commit 2cf63edc44e8b2c467edb41e75ea0727462254f4 (staging: unisys: visornic: Remove errant -EIO returns).

davidker, 2017-01-13:
Fixed in forlater

closed 2017-01-13 by davidker