davidker/unisys #107: KB-25100

TurningWrenches, 2016-10-28:
[KanBoard-25100](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=25100&project_id=377)

selltc, 2016-10-28:
removed an extraneous "7" from the KanBoard link:
http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=25100&project_id=377
-->
http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=25100&project_id=37

TurningWrenches, 2016-10-28:
**Branch:** githubissue-107-for-later-TurningWrenches
**Commit(s):** 

**checkpatch-cleared:** Yes
**T710-1 verified:** No

selltc, 2016-11-06:
I successfully verified David's fix in the [githubissue-106-for-later-TurningWrenches branch](/davidker/unisys/commits/githubissue-106-for-later-TurningWrenches) (which I fitted into the [githubissue-106-for-later-201609131535 branch](/davidker/unisys/commits/githubissue-106-for-later-201609131535) for my testing) on my T110 as described [in comment-13 of Trac-7233](https://ustr-svn-1.na.uis.unisys.com/trac/spar/ticket/7233#comment:13) and [in comment-23 of Trac-7234](https://ustr-svn-1.na.uis.unisys.com/trac/spar/ticket/7234#comment:23).  **So it appears we DID get the "2-for-1 deal", in that the fix for #106 also fixes #107.**

Note I also needed commit b1359d11b381ce4aedae021eb5154425872fe64a (remove unnecessary declaration of visorbus_free_irq()) in order to compile cleanly.

TurningWrenches, 2016-11-22:
Thanks, Tim! I appended that same revision (from my own branch) in GitHub issue #107 (fc44a49d292a00284972f79b7c83d378727bf5f1).

selltc, 2016-11-29:
We also need to add commit 34801316e9131ad2ac05dde83e5ec1f8c179e2c4 (staging: unisys: remove redundant unlocks of visornic_devdata.priv_lock) here, in order to fix a problem introduced by commit 2cf63edc44e8b2c467edb41e75ea0727462254f4 (staging: unisys: visornic: Remove errant -EIO returns).

davidker, 2017-01-12:
@selltc can we close these or should we wait u til we have patches squashed?

selltc, 2017-01-12:
If the fix patches from [githubissue-106-for-later-201609131535 branch](/davidker/unisys/commits/githubissue-106-for-later-201609131535) and commit 34801316e9131ad2ac05dde83e5ec1f8c179e2c4 (staging: unisys: remove redundant unlocks of visornic_devdata.priv_lock) are in for-later, we should close this.

closed 2017-01-12 by selltc