davidker/unisys #108: Don't check for more than PAGE_SIZE length

TurningWrenches, 2016-10-31:


TurningWrenches, 2016-10-31:
Just so we know what we're working with:

```
$ grep -rn "scnprintf" drivers/staging/unisys/*
drivers/staging/unisys/visorbus/visorchipset.c:191:     return scnprintf(buf, PAGE_SIZE, "%u\n", tool_action);
drivers/staging/unisys/visorbus/visorchipset.c:226:     return scnprintf(buf, PAGE_SIZE, "%u\n",
drivers/staging/unisys/visorbus/visorchipset.c:262:     return scnprintf(buf, PAGE_SIZE, "%i\n", error);
drivers/staging/unisys/visorbus/visorchipset.c:295:     return scnprintf(buf, PAGE_SIZE, "%i\n", text_id);
drivers/staging/unisys/visorbus/visorchipset.c:327:     return scnprintf(buf, PAGE_SIZE, "%hu\n", remaining_steps);
```

There are other ```scnprintf's``` in our driver code, but the instances above seem to be the only ones that relate to this issue.

TurningWrenches, 2016-10-31:
**Branch:** githubissue-108-for-later-TurningWrenches
**Commit(s):** 
5b927c36c18e8359398b12eff31a3a5c99fc3144
**checkpatch-cleared:** Yes
**T710-1 verified:** No

davidker, 2016-11-10:
Can you split this into two patches? One for each file? I would like to put the Reported-by on the one for visorchipset, not the one for visorbus.

TurningWrenches, 2016-11-10:
**Branch:** githubissue-108-upstream-next-TurningWrenches
**Commit(s):** 
3f185e5095c76fb405f25233f139b69c5f45370f
6aa7db3f392170089596e952e525a31ff3322ba8
**checkpatch-cleared:** Yes
**T710-1 verified:** No

davidker, 2017-01-12:
This patch was accepted into the upstream. Closing.

closed 2017-01-12 by davidker