davidker/unisys #109: Use case statement for crash enum

TurningWrenches, 2016-11-01:
``` 
 > 		if (visorchannel_write(controlvm_channel,
 > 				       local_crash_msg_offset,
 > 				       msg,
 > 				       sizeof(struct controlvm_message)) < 0) {
 > 			POSTCODE_LINUX_2(SAVE_MSG_BUS_FAILURE_PC,
 > 					 POSTCODE_SEVERITY_ERR);
 > 			return;
 > 		}
 > 	} else {
 > 		local_crash_msg_offset += sizeof(struct controlvm_message);
 > 		if (visorchannel_write(controlvm_channel,
 > 				       local_crash_msg_offset,
 > 				       msg,
 > 				       sizeof(struct controlvm_message)) < 0) {
 > 			POSTCODE_LINUX_2(SAVE_MSG_DEV_FAILURE_PC,
 > 					 POSTCODE_SEVERITY_ERR);
 > 			return;
 > 		}
 ```
 So an enum yet you don't check the other value?  case statement perhaps?

TurningWrenches, 2016-11-01:
**Branch:** githubissue-109-for-later-TurningWrenches
**Commit(s):** 
dc9a26b8414b7a67cc907b75469323c8daa77930
**checkpatch-cleared:** Yes
**T710-1 verified:** No

davidker, 2016-12-01:
Was this built off of for-later?

TurningWrenches, 2016-12-01:
That should say "upstream-next". Sorry about the typo :(

davidker, 2017-01-17:
Closed committed at 36309d3b105c650eb0398afd1675df30fb570e5d

closed 2017-01-17 by davidker