davidker/unisys #111: Clean up retry variable

TurningWrenches, 2016-11-04:
[KanBoard-24501](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=24501&project_id=37)

```
 > 
 > static struct parser_context *
 > parser_init_byte_stream(u64 addr, u32 bytes, bool local, bool *retry)
 > {
 > 	int allocbytes = sizeof(struct parser_context) + bytes;
 > 	struct parser_context *ctx;
 > 
 > 	if (retry)
 > 		*retry = false;
 ```

 Why would retry be NULL?  Doesn't look like you ever have it set that
 way, so why check it?
 
```
 > 
 > 	/*
 > 	 * alloc an 0 extra byte to ensure payload is
 > 	 * '\0'-terminated
 > 	 */
 > 	allocbytes++;
 > 	if ((controlvm_payload_bytes_buffered + bytes)
 > 	    > MAX_CONTROLVM_PAYLOAD_BYTES) {
 > 		if (retry)
 > 			*retry = true;
 > 		return NULL;
 > 	}
 > 	ctx = kzalloc(allocbytes, GFP_KERNEL | __GFP_NORETRY);
 ```

 Why no retry?

TurningWrenches, 2016-11-04:
Just so we know what we're dealing with:

```
$ grep -rn "retry" drivers/staging/unisys/
drivers/staging/unisys/visorbus/visorchipset.c:1828:parser_init_byte_stream(u64 addr, u32 bytes, bool local, bool *retry)
drivers/staging/unisys/visorbus/visorchipset.c:1833:    if (retry)
drivers/staging/unisys/visorbus/visorchipset.c:1834:            *retry = false;
drivers/staging/unisys/visorbus/visorchipset.c:1843:            if (retry)
drivers/staging/unisys/visorbus/visorchipset.c:1844:                    *retry = true;
drivers/staging/unisys/visorbus/visorchipset.c:1849:            if (retry)
drivers/staging/unisys/visorbus/visorchipset.c:1850:                    *retry = true;
drivers/staging/unisys/visorbus/visorchipset.c:1923:            bool retry = false;
drivers/staging/unisys/visorbus/visorchipset.c:1927:                                        local_addr, &retry);
drivers/staging/unisys/visorbus/visorchipset.c:1928:            if (!parser_ctx && retry)
drivers/staging/unisys/visornic/visornic_main.c:1513:    * but still retry it if you cant allocate them all this time.
drivers/staging/unisys/include/channel.h:325:* retry policy (e.g. wait & try again, report an error, etc.).
```

TurningWrenches, 2016-12-02:
**Branch:** githubissue-111-upstream-next-TurningWrenches
**Commit(s):** 
91747dac47830bd0cc395f697bc6b148425d60e7
**checkpatch-cleared:** Yes
**T710-W1 verified:**

davidker, 2017-01-12:
Commited at 3e4273d

closed 2017-01-12 by davidker