davidker/unisys #113: Rename typ to type

TurningWrenches, 2016-12-02:
```
 > 	}
 > 
 > 	if (visorchannel_read(controlvm_channel,
 > 			      offsetof(struct spar_controlvm_channel_protocol,
 > 				       saved_crash_message_offset),
 > 			      &local_crash_msg_offset, sizeof(u32)) < 0) {
 > 		POSTCODE_LINUX_2(CRASH_DEV_CTRL_RD_FAILURE_PC,
 > 				 POSTCODE_SEVERITY_ERR);
 > 		return;
 > 	}
 > 
 > 	if (typ == CRASH_BUS) {
 ```
 here, have a 'e', it's cheap...

davidker, 2016-12-02:
Please don't ;)

TurningWrenches, 2016-12-02:
Just for future reference:

```
$ grep -rnw "typ" drivers/staging/unisys
drivers/staging/unisys/visorbus/visorchipset.c:584:save_crash_message(struct controlvm_message *msg, enum crash_obj_type typ)
drivers/staging/unisys/visorbus/visorchipset.c:619:     switch (typ) {
drivers/staging/unisys/include/channel.h:217:#define spar_signal_init(chan, QHDRFLD, QDATAFLD, QDATATYPE, ver, typ)     \
drivers/staging/unisys/include/channel.h:221:           chan->QHDRFLD.chtype = typ;    
```

davidker, 2016-12-02:
Due to several internal tools, I don't think we can use the word "type" so we will need to find something else or be more specific in the variable name for "type".

closed 2017-07-12 by davidker