davidker/unisys #114: Add error handling for crash_message errors

TurningWrenches, 2016-12-02:
```
 > 
 > static void
 > save_crash_message(struct controlvm_message *msg, enum crash_obj_type typ)
 > {
 > 	u32 local_crash_msg_offset;
 > 	u16 local_crash_msg_count;
 > 
 > 	if (visorchannel_read(controlvm_channel,
 > 			      offsetof(struct spar_controlvm_channel_protocol,
 > 				       saved_crash_message_count),
 > 			      &local_crash_msg_count, sizeof(u16)) < 0) {
 > 		POSTCODE_LINUX_2(CRASH_DEV_CTRL_RD_FAILURE_PC,
 > 				 POSTCODE_SEVERITY_ERR);
 > 		return;
 > 	}
 > 
 > 	if (local_crash_msg_count != CONTROLVM_CRASHMSG_MAX) {
 > 		POSTCODE_LINUX_3(CRASH_DEV_COUNT_FAILURE_PC,
 > 				 local_crash_msg_count,
 > 				 POSTCODE_SEVERITY_ERR);
 > 		return;
 ```
 No errors being returned despite this obviously being an error?

TurningWrenches, 2016-12-02:
Started by just finding all instances of "crash_msg":

```
$ grep -rn "crash_msg" drivers/staging/unisys
drivers/staging/unisys/visorbus/controlvmchannel.h:436:  struct controlvm_message saved_crash_msg[CONTROLVM_CRASHMSG_MAX];
drivers/staging/unisys/visorbus/controlvmchannel.h:457: offsetof(struct spar_controlvm_channel_protocol, saved_crash_msg)
drivers/staging/unisys/visorbus/visorchipset.c:586:     u32 local_crash_msg_offset;
drivers/staging/unisys/visorbus/visorchipset.c:587:     u16 local_crash_msg_count;
drivers/staging/unisys/visorbus/visorchipset.c:593:                             &local_crash_msg_count, sizeof(u16));
drivers/staging/unisys/visorbus/visorchipset.c:601:     if (local_crash_msg_count != CONTROLVM_CRASHMSG_MAX) {
drivers/staging/unisys/visorbus/visorchipset.c:604:                                   local_crash_msg_count, DIAG_SEVERITY_ERR);
drivers/staging/unisys/visorbus/visorchipset.c:611:                             &local_crash_msg_offset, sizeof(u32));
drivers/staging/unisys/visorbus/visorchipset.c:621:             local_crash_msg_offset += sizeof(struct controlvm_message);
drivers/staging/unisys/visorbus/visorchipset.c:623:                                      local_crash_msg_offset,
drivers/staging/unisys/visorbus/visorchipset.c:635:                                      local_crash_msg_offset,
drivers/staging/unisys/visorbus/visorchipset.c:1586:    u32 local_crash_msg_offset;
drivers/staging/unisys/visorbus/visorchipset.c:1587:    u16 local_crash_msg_count;
drivers/staging/unisys/visorbus/visorchipset.c:1603:                          &local_crash_msg_count, sizeof(u16)) < 0) {
drivers/staging/unisys/visorbus/visorchipset.c:1610:    if (local_crash_msg_count != CONTROLVM_CRASHMSG_MAX) {
drivers/staging/unisys/visorbus/visorchipset.c:1613:                                  local_crash_msg_count, DIAG_SEVERITY_ERR);
drivers/staging/unisys/visorbus/visorchipset.c:1621:                          &local_crash_msg_offset, sizeof(u32)) < 0) {
drivers/staging/unisys/visorbus/visorchipset.c:1630:                          local_crash_msg_offset,
drivers/staging/unisys/visorbus/visorchipset.c:1641:                          local_crash_msg_offset +
```

TurningWrenches, 2016-12-02:
Issue fixed in commit 8dff01f7d.

davidker, 2016-12-02:
Sorry, a lot of the error ones are fixed due to the work I'm doing in the overall error handling task. I guess we can clean some of this up.

closed 2017-02-28 by davidker