davidker/unisys #116: Explain logic in bus_destroy()

TurningWrenches, 2016-12-05:
```
 > 	}
 > 
 > 	INIT_LIST_HEAD(&bus_info->list_all);
 > 	bus_info->chipset_bus_no = bus_no;
 > 	bus_info->chipset_dev_no = BUS_ROOT_DEVICE;
 > 
 > 	POSTCODE_LINUX_3(BUS_CREATE_ENTRY_PC, bus_no, POSTCODE_SEVERITY_INFO);
 > 
 > 	visorchannel = visorchannel_create(cmd->create_bus.channel_addr,
 > 					   cmd->create_bus.channel_bytes,
 > 					   GFP_KERNEL,
 > 					   cmd->create_bus.bus_data_type_uuid);
 > 
 > 	if (!visorchannel) {
 > 		POSTCODE_LINUX_3(BUS_CREATE_FAILURE_PC, bus_no,
 > 				 POSTCODE_SEVERITY_ERR);
 > 		rc = -CONTROLVM_RESP_ERROR_KMALLOC_FAILED;
 > 		kfree(bus_info);
 > 		bus_info = NULL;
 > 		goto out_bus_epilog;
 > 	}
 > 	bus_info->visorchannel = visorchannel;
 > 	if (uuid_le_cmp(cmd->create_bus.bus_inst_uuid, spar_siovm_uuid) == 0) {
 > 		dump_vhba_bus = bus_no;
 > 		save_crash_message(inmsg, CRASH_BUS);
 > 	}
 > 
 > 	POSTCODE_LINUX_3(BUS_CREATE_EXIT_PC, bus_no, POSTCODE_SEVERITY_INFO);
 > 
 > out_bus_epilog:
 > 	bus_epilog(bus_info, CONTROLVM_BUS_CREATE, &inmsg->hdr,
 > 		   rc, inmsg->hdr.flags.response_expected == 1);
 > }
 > 
 > static void
 > bus_destroy(struct controlvm_message *inmsg)
 > {
 > 	struct controlvm_message_packet *cmd = &inmsg->cmd;
 > 	u32 bus_no = cmd->destroy_bus.bus_no;
 > 	struct visor_device *bus_info;
 > 	int rc = CONTROLVM_RESP_SUCCESS;
 > 
 > 	bus_info = visorbus_get_device_by_id(bus_no, BUS_ROOT_DEVICE, NULL);
 > 	if (!bus_info)
 > 		rc = -CONTROLVM_RESP_ERROR_BUS_INVALID;
 ```
 How can this happen?

TurningWrenches, 2016-12-07:
**Branch:** githubissue-116-upstream-next-TurningWrenches
**Commit(s):** 
480a9e38c42bccbeb33057b2815c7048cf83b6e2
**checkpatch-cleared:** Yes
**T710-W1 verified:**

davidker, 2017-03-10:
Tim sent email after patch was denied.

closed 2017-03-10 by davidker