davidker/unisys #117: kobject function names

TurningWrenches, 2016-12-08:
[See Kanboard 24549](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=24549&project_id=37)
```
 > 	if (msg_hdr->flags.response_expected)
 > 		controlvm_respond(msg_hdr, rc);
 ```
 And your kobject function names stink, make them mean what they do.

TurningWrenches, 2016-12-08:
**Branch:** githubissue-117-staging-next-TurningWrenches
**Commit(s):** 
6888bae97510d1886a6b5fe68b325986ea0dbf42
**checkpatch-cleared:** Yes
**T710-1 verified:**

TurningWrenches, 2017-01-20:
I ran with patch 5c426993b9cf8d0e8089f0ea1e0e3b8c0662de51 on a copy of this branch. On the machine I tested with, I installed a custom udev rule file for rootfs and the initramfs:
```
$ more /etc/udev/rules.d/99-spar_guest.rules
# Copyright (c) 2013 - 2016 UNISYS CORPORATION
# All rights reserved
#
# These are s-Par-specific rules for udev.

DEVPATH=="/devices/platform/visorchipset*" , ACTION=="add" , RUN+="/lib/udev/spar_visorchipset_guest.sh"
DEVPATH=="/devices/platform/visorchipset*" , ACTION=="online" , RUN+="/lib/udev/spar_visorchipset_guest.sh"
DEVPATH=="/devices/platform/visorchipset*" , ACTION=="offline" , RUN+="/lib/udev/spar_visorchipset_guest.sh"
DEVPATH=="/devices/platform/visorchipset*" , ACTION=="change" , RUN+="/lib/udev/spar_visorchipset_guest.sh"
```

The same for the referenced script:
```
$ more /lib/udev/spar_visorchipset_guest.sh
#!/bin/sh
#
# Copyright (c) 2013 - 2016 UNISYS CORPORATION
# All rights reserved.

if [ "$ACTION" = "add" ]
then
        if [ "$SPAR_PARAHOTPLUG" = 1 ]
        then
                /sbin/spar_parahotplug.sh
        fi

        logger "GGG: Received kobj_uevent - ADD (rootfs)"
fi

if [ "$ACTION" = "online" ]
then
        if [ "$SPAR_PARAHOTPLUG" = 1 ]
        then
                /sbin/spar_parahotplug.sh
        fi

        logger "GGG: Received kobj_uevent - ONLINE (rootfs)"
fi

if [ "$ACTION" = "offline" ]
then
        if [ "$SPAR_PARAHOTPLUG" = 1 ]
        then
                /sbin/spar_parahotplug.sh
        fi

        logger "GGG: Received kobj_uevent - OFFLINE (rootfs)"
fi

if [ "$ACTION" = "change" ]
then
        if [ "$SPAR_PARAHOTPLUG" = 1 ]
        then
                /sbin/spar_parahotplug.sh
        fi

        logger "GGG: Received kobj_uevent - CHANGE (rootfs)"
fi
```

(Just substitute 'initramfs' for 'rootfs' for the initramfs version).

The results were as expected:
```
Jan 19 19:21:24 r3201 spar: GGG - 11917 TEST
Jan 19 19:21:49 r3201 logger: GGG: Received kobj_uevent - ADD (initramfs)
Jan 19 19:21:50 r3201 logger: GGG: Received kobj_uevent - ADD (initramfs)
Jan 19 19:21:50 r3201 logger: GGG: Received kobj_uevent - OFFLINE (initramfs)
Jan 19 19:21:50 r3201 logger: GGG: Received kobj_uevent - ONLINE (initramfs)
Jan 19 19:21:50 r3201 logger: GGG: Received kobj_uevent - CHANGE (initramfs)
Jan 19 19:21:58 r3201 logger: GGG: Received kobj_uevent - ADD (rootfs)
```

davidker, 2017-02-06:
Committed to staging-next at 7289a8d

closed 2017-02-06 by davidker